OPCFoundation/UA-.NETStandard #770: how to make local server  - client connection, pc to pc, client name in loca but server name is not

SShaarawy, 2019-07-26:
we executed Universal windows.sln but we cannot

MD-V, 2019-07-28:
Hello @SShaarawy,

please elaborate your issue little more.

commcad, 2019-07-30:
He means (I think) his server's endpoint is a FQDN and I suspect his client can't connect to it. Therefore more or less a local networking/routing issue. Please DM if I win anything...

SShaarawy, 2019-07-31:
Firstly, we looked to sampleserver and sampleclient programs to if they see eachother. 
In cmd, they see but in windows form they can not. Client looking for server in Localhost however server is looking client in different area, we understood like that. We are trying to in windows form 
make a connection. Thanks for your returns. Also my main aim is connect our sampleclient to Matrikon OPC server. Do you know something about that ? let me know.

MD-V, 2019-07-31:
Hello @SShaarawy,

I'm still not sure if I understand you correctly.
I have created an example server and an example client. You can find them here:
[Client](https://gist.github.com/MD-V/275e89532351f1777ddcec3bf4e7ac99)
[Server](https://gist.github.com/MD-V/4067d5386008731e5e4585cec62a2cdc)
Server is running on localhost. Client searches directly on the localhost.

Please try to put the endpoint url of the matrikon server in the Client Example (adjust line 172 with an exisiting NodeId and try to connect)

mregen, 2019-07-31:
@SShaarawy In Universal Windows there is a feature called loopback network isolation, the Universal Windows client / server cannot connect by default to each other. Please run them on different machines for testing or use the .Net 4.6 versions.

mregen, 2019-07-31:
There was somebody seeing this problem here too: https://github.com/OPCFoundation/UA-.NETStandard/issues/440

SShaarawy, 2019-08-02:
thank you mregen, MD-V,commcad

SShaarawy, 2019-08-02:
we connected cleint and server, server from cmd and client from windows solution. that is working now.

MD-V, 2019-08-06:
@SShaarawy 

please close the issue if resolved :+1:

closed 2019-08-06 by SShaarawy