OPCFoundation/UA-.NETStandard #772: Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid

KmanViel, 2019-07-31:
Having an issue with a Xamarin Android app where it is throwing this certificate exception:  Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid

I am not familiar with the certificate that this code base is looking for as I am a newbie.  The application worked for quite some time (many months) and now it throws this exception.

When running in Visual Studio and debugging on the Android device the first exception happens in CertificateFactory.GetKeyUsage when cert.Extensions.OfType<X509KeyUsageExtension>() is called (method below).

    public static X509KeyUsageFlags GetKeyUsage(X509Certificate2 cert)
    {
        X509KeyUsageFlags allFlags = X509KeyUsageFlags.None;
        foreach (X509KeyUsageExtension ext in cert.Extensions.OfType<X509KeyUsageExtension>())
        {
            allFlags |= ext.KeyUsages;
        }
        return allFlags;
    } 

How do I get around the certificate exception?  Any other information about how the certificate works would be welcome.

MD-V, 2019-07-31:
@KmanViel,

is the own self signed cert expired or the cert of the server?

KmanViel, 2019-07-31:
It must be a certificate on the Android device itself because the app used to run without connecting to the server via WiFi but it no longer does.  Below is a section of the application's config file showing where the certificates are located.  The certificates in the "own" and "trusted" folders have expired.  I am a newbie on certificates.  How are the valid dates changed?  Can the application run without a certificate or a one that does not expire? Where can I find documentation on certificates?

From the application's config file:
  <SecurityConfiguration>

    
    <ApplicationCertificate>
      <StoreType>Directory</StoreType>
      <StorePath>/storage/emulated/0/OPC Foundation/PKI/own</StorePath>
      <SubjectName>CN=UA Xamarin Sample Client, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
    </ApplicationCertificate>

    
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>/storage/emulated/0/OPC Foundation/PKI/issuer</StorePath>
    </TrustedIssuerCertificates>

    
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>/storage/emulated/0/OPC Foundation/PKI/trusted</StorePath>
    </TrustedPeerCertificates>

    
    <RejectedCertificateStore>
      <StoreType>Directory</StoreType>
      <StorePath>/storage/emulated/0/OPC Foundation/PKI/rejected</StorePath>
    </RejectedCertificateStore>

    
    <AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>

  </SecurityConfiguration>

mregen, 2019-07-31:
Hi @KmanViel, if you can delete the expired app cert in the 'own' folder it should create a new self signed cert on the next start, again valid for a year or so.

KmanViel, 2019-07-31:
Is it possible to have this done automatically or not use a cert?

MD-V, 2019-08-01:
@KmanViel,

the method ApplicationInstance.CheckApplicationInstanceCertificate checks if the cert is valid. If its not you could implement a method similar to ApplicationInstance.DeleteApplicationInstanceCertificate to delete the invalid cert
(Reload the ApplicationConfig after deleting the cert).
After you deleted the cert you can call ApplicationInstance.CheckApplicationInstanceCertificate again and it'll create a new self signed application cert for you.

KmanViel, 2019-08-01:
@mregen @MD-V 
Thank you both for the information!

MD-V, 2019-08-06:
@KmanViel 

please close the issue if resolved :+1:

closed 2019-08-06 by KmanViel

yachtingchen, 2023-11-09:
Is there a way to update a certificate when it's about to expire without shutting down the server? This is important for services that must operate continuously without downtime.