OPCFoundation/UA-.NETStandard #774: How to parse data which is sending from client to server?

SShaarawy, 2019-08-02:
we made a connection between client-server but we could not show the item that is sending from client to cmd screen.
We executed server part from cmd and client part from windows.sln (visual studio), we are counting the sent items on cmd screen but we can not show them. How can we do that, do you have any suggestions?

MD-V, 2019-08-03:
Hello @SShaarawy,

could you please be more specific? What are you trying to do? 
Value change notification? Event? Methods?

SShaarawy, 2019-08-05:
we are trying to do show the data which is sending from client to server . When we execute the sampleclient, we see the objects in windows form. Program is sending to server side when we click report button. We are counting sent objects with calling currentMonitoredItemsCount. But we cannot showing them.

MD-V, 2019-08-06:
Hello @SShaarawy,

I'm unable to help you because your message doesnt contain many details. 
Which Projects in the Sample Solution are you using?

Are you trying to read events from a server?

MD-V, 2019-08-15:
@SShaarawy Please close the issue if resolved.

closed 2019-08-22 by AlinMoldovean