OPCFoundation/UA-.NETStandard #775: to get multiple nodes at once

hitomi12345, 2019-08-05:
Hello
I implemented a Client to get the value of 700 Veribable class nodes from the Server.
It is an implementation to call the ReadValue (NodeId nodeId) method in the Session class of the SDK 700 times to get the value.
　　　　
```
            String contents = "";
            for (int i = 0; i < 700; i++)
            {
                contents += m_session.ReadValue("ns=2;s=Id" + i.ToString().PadLeft(4, '0')).ToString() + ",";
            }
```

However, it takes more than 10 seconds to get it, which is too late.
I think that the reason why it takes so much time is because the Server and Client communicate 700 times.

Is there any way to get multiple nodes at once?
And is there any other way to improve speed?

Thank you.

MD-V, 2019-08-05:
Hello @hitomi12345,

you could use the method Session.ReadValues [(see here)](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/SDK/Opc.Ua.Client/Session.cs#L2743) which only calls the server once.

MD-V, 2019-08-15:
@hitomi12345 Please close the issue if resolved.

closed 2019-08-22 by AlinMoldovean