OPCFoundation/UA-.NETStandard #777: registe to global discovery server , ***EXCEPTION*** BadServiceUnsupported ''i=12193' is an unrecognized

shirenfeng, 2019-08-06:
but ,defined  : 
        /// <summary>
        /// The identifier for the RegisterServer2Request DataType.
        /// </summary>
        public const uint RegisterServer2Request = 12193;

2019/8/6 14:56:50.301 ApplicationAdmin Token Accepted: appadmin
2019/8/6 14:56:50.302 Session FIRST ACTIVATION, Id=ns=4;i=762993770, Name=Session i=0, ChannelId=270ca12d-67e1-456f-9e2d-de89bfc71654-45, User=appadmin
2019/8/6 14:56:50.303 Server - SESSION ACTIVATED.
2019/8/6 14:56:50.304 Channel 45: SendResponse 3
2019/8/6 14:56:50.816 Channel 45: ProcessRequestMessage 4
2019/8/6 14:56:50.817 ***EXCEPTION*** BadServiceUnsupported ''i=12193' is an unrecognized service identifier.' 'i=12193' is an unrecognized service identifier.
2019/8/6 14:56:50.818 Channel 45: SendResponse 4
2019/8/6 14:56:50.819 Channel 45: ProcessRequestMessage 5
2019/8/6 14:56:50.820 ***EXCEPTION*** BadServiceUnsupported ''i=435' is an unrecognized service identifier.' 'i=435' is an unrecognized service identifier.
2019/8/6 14:56:50.820 Channel 45: SendResponse 5
2019/8/6 14:56:51.823 Channel 45: ProcessRequestMessage 6
2019/8/6 14:56:51.826 ***EXCEPTION*** BadServiceUnsupported ''i=12193' is an unrecognized service identifier.' 'i=12193' is an unrecognized service identifier.
2019/8/6 14:56:51.829 Channel 45: SendResponse 6
2019/8/6 14:56:51.834 Channel 45: ProcessRequestMessage 7
2019/8/6 14:56:51.837 ***EXCEPTION*** BadServiceUnsupported ''i=435' is an unrecognized service identifier.' 'i=435' is an unrecognized service identifier.
2019/8/6 14:56:51.840 Channel 45: SendResponse 7

AlinMoldovean, 2019-08-13:
@shirenfeng ,

It seems that you have an older version of Local Discovery Server without support for RegisterServer2.

shirenfeng, 2019-08-15:
@AlinMoldovean ,
the master version:
<VersionPrefix>1.04.354</VersionPrefix>

AlinMoldovean, 2019-08-22:
@shirenfeng ,

The problem seems to be that you invoke a RegisterServer2 service call against a server developed with this UA .NET Stack.
This service should only be invoked against a LDS (Local Discovery Server) installation.

shirenfeng, 2019-08-26:
@AlinMoldovean ,

I use open62541's API to registe to GDS:
...
UA_ServerConfig_setDefaultWithSecurityPolicies(..)
...
UA_Server_addPeriodicServerRegisterCallback(..)

AlinMoldovean, 2019-08-29:
The RegisterServer2  is not intended to be supported by a normal UA Server.
It is only intended for LDS!

closed 2019-08-29 by AlinMoldovean