OPCFoundation/UA-.NETStandard #778: Datatype of (User)AccessLevel is now uint32 in NodeSet

SebastianBaeumler, 2019-08-06:
The datatype of attribute (User)AccessLevel in the XML NodeSet has been changed from byte to uint32 in Version 1.04 of the OPC UA specification.
This change in the schema file has not been reflected in the source code:

**UANodeSet.xsd:**
        
        <xs:simpleType name="AccessLevel">
             <xs:restriction base="xs:unsignedInt"></xs:restriction>
        </xs:simpleType>

        <xs:attribute name="AccessLevel" type="AccessLevel" default="1"></xs:attribute>
        <xs:attribute name="UserAccessLevel" type="AccessLevel" default="1"></xs:attribute>

**UANoteSet.cs:**

        private byte accessLevelField;
        private byte userAccessLevelField;
        /// <remarks/>
        [System.Xml.Serialization.XmlAttributeAttribute()]
        [System.ComponentModel.DefaultValueAttribute(typeof(byte), "1")]
        public byte AccessLevel {
            get {
                return this.accessLevelField;
            }
            set {
                this.accessLevelField = value;
            }
        }
        /// <remarks/>
        [System.Xml.Serialization.XmlAttributeAttribute()]
        [System.ComponentModel.DefaultValueAttribute(typeof(byte), "1")]
        public byte UserAccessLevel {
            get {
                return this.userAccessLevelField;
            }
            set {
                this.userAccessLevelField = value;
            }
        }

SebastianBaeumler, 2019-10-02:
#815

AlinMoldovean, 2019-10-16:
Closed with #815

closed 2019-10-16 by AlinMoldovean