OPCFoundation/UA-.NETStandard #782: How to read the description of a node?

SvenSchlesinger, 2019-08-14:
Hi!

I implemented a client, which browses through a server and retrieves all information from the nodes (like the DisplayName, NodeClass etc.). It works fine.

When I am using the UaExpert, it is also possible to read the "Description" attribute of a Node.
Unfortunately, I did not find a solution to read this attribute while using UA-.NETStandard.

Is there any possibility to read the description attribute of a Node?

Thanks!

AlinMoldovean, 2019-08-14:
Hi @SvenSchlesinger ,

You can read the Description attribute of a node by using the _SessionClient.Read()_ method:

```
public virtual ResponseHeader Read(
            RequestHeader                requestHeader,
            double                       maxAge,
            TimestampsToReturn           timestampsToReturn,
            ReadValueIdCollection        nodesToRead,
            out DataValueCollection      results,
            out DiagnosticInfoCollection diagnosticInfos)
```

A good example can found in the implementation of the _Session.ReadValue()_ method.
You only need to change the _AttributeId_ from Value to _Description_.
Something like this:

```
        public DataValue ReadDescription(NodeId nodeId)
        {
            ReadValueId itemToRead = new ReadValueId();

            itemToRead.NodeId = nodeId;
            itemToRead.AttributeId = Attributes.Description;

            ReadValueIdCollection itemsToRead = new ReadValueIdCollection();
            itemsToRead.Add(itemToRead);

            // read from server.
            DataValueCollection values = null;
            DiagnosticInfoCollection diagnosticInfos = null;

            ResponseHeader responseHeader = Read(
                null,
                0,
                TimestampsToReturn.Both,
                itemsToRead,
                out values,
                out diagnosticInfos);

            ClientBase.ValidateResponse(values, itemsToRead);
            ClientBase.ValidateDiagnosticInfos(diagnosticInfos, itemsToRead);

            if (StatusCode.IsBad(values[0].StatusCode))
            {
                ServiceResult result = ClientBase.GetResult(values[0].StatusCode, 0, diagnosticInfos, responseHeader);
                throw new ServiceResultException(result);
            }

            return values[0];
        }
```

SvenSchlesinger, 2019-08-14:
Thank your very much @AlinMoldovean ! That helped me a lot!

closed 2019-08-29 by AlinMoldovean