OPCFoundation/UA-.NETStandard #783: Unable to connect from Linux to Windows OPC UA server

ViXXoR, 2019-08-14:
When trying to connect to a KEPServerEXv6 server from a Raspberry Pi (Raspbian Buster), I'm getting the following error:
`Opc.Ua.ServiceResultException: Error establishing a connection: Error received from remote host: An error occurred verifying security.`

Full stack trace:
```
Unhandled Exception: System.AggregateException: One or more errors occurred. (Error establishing a connection: Error received from remote host: An error occurred verifying security.) ---> Opc.Ua.ServiceResultException: Error establishing a connection: Error received from remote host: An error occurred verifying security.
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
```

- I see the client appear in the list within the KEPServer configuration screen
- I have accepted the client certificate on the server
- I have placed the server certificate in the applications trusted folder

Nothing I do seems to get me closer to fixing this issue.

I have been fighting with this for a couple of days and am at my wits end.

MD-V, 2019-08-15:
@ViXXoR,

are there logs from the KEPServerEXv6? Because the error is on server side. Maybe unsupported security config or an error in the client certificate?

ViXXoR, 2019-08-15:
@MD-V,

Unfortunately the event log on KEPServer shows no security or other errors when I try to connect.

I'm connecting from a Windows machine using a certificate created with the same 
`CertificateFactory.CreateCertificate(...)` function. So both certificates are 2048 RSA PKs with SHA256 hash.

Is there any tool I might be able to use to verify the certificates?
I guess I should also look into contacting KEPServer support.

ViXXoR, 2019-08-17:
In the end, this turned out to be a misconfiguration/misunderstanding on my part.

Thank you for your help.

closed 2019-08-17 by ViXXoR

s7788, 2020-03-30:
Can you share how to solve it?

thanks a lot

ViXXoR, 2020-03-30:
Turned out I was attempting to connect to two different servers in code, and one has not accepted my certificate.

You may also want to ensure you enable the automatic acceptance of certificates in the client code.