OPCFoundation/UA-.NETStandard #785: Xamarin demo client for Android hangs on connect

trofimich, 2019-08-28:
I've compiled original "UA Xamarin Client" solution and tried to run it on my Android 8.1 device and found that i can't connect to OPC-UA server. Application simply hangs on connection attempt.

During debuggging i found thatthe problem is in file Opc.Ua.Core/Security/Certificates/DirectoryCertificateStore.cs line 613:

`entry.Certificate = new X509Certificate2(file.FullName);`

This line of code never finishes. I tried to comment this code, but it hangs further (also when trying to make some manipulations with certificates).

So, the sample is completely broken :-/

wsaiho, 2019-10-16:
I had to overcome 2 issues before being able to connect to the OPC-UA server (ICE3-8IOL-G65L-V1D) with the Xamarin client.

1. Access to the path "/storage/emulated/0/Opc.Ua.SampleClient.Config.xml" is denied
It turns out that I need to give the [app permission](https://stackoverflow.com/questions/46911486/xamarin-system-unauthorizedaccessexception-access-to-the-path-is-denied), so I copied and paste the CheckAppPermissions() function to MainActivity.cs. I only had to give permission once when the app runs for the first time.

2. In SecurityPolicies.cs, there is a GetDisplayName method that get invoked. Within it, the GetFields method returns nothing. I am not sure how to solve this, but I know that the security policy is None for my OPC-UA server so I just hardcoded the return "None" for this GetDisplayName method.

ttmmduy, 2019-10-19:
I've added CheckAppPermissions(), the previuos issue has been resolved, but then I receive a warning "cannot connect to an OPC UA server"

wsaiho, 2019-10-19:
You step through the code to see if the exception happens at the same place as issue #2 in my previous post?

ttmmduy, 2019-10-20:
thanks for advice, i connected the server, but now i can not read or write the values, do you know why?

wsaiho, 2019-10-20:
Can you at least see the whole browse tree and the nodes?

ttmmduy, 2019-10-20:
[image]

wsaiho, 2019-10-20:
It happened to me to. I can see the browse tree of my OPC-UA server but cannot read the values at each node. I am not sure why. If you manage to solve it please share how :)