OPCFoundation/UA-.NETStandard #786: Latest Nuget Package

kieranhaycock, 2019-08-29:
Hi all,

I've been using the latest Nuget package and it seems a few months out of date. Any news on when a new version will be released? I would prefer to use the Nuget package and not the source but the samples are not fully compatible with the latest version.

Thanks all.

kieranhaycock, 2019-08-29:
The main issue is it looks as though MonitoredNode has been replaced with MonitoredNode2. However the constructor takes in a CustomNodeManager2.

AlinMoldovean, 2019-08-29:
The latest release of  Nuget package compiled from this repository is one month old and should be compatible with the actual code base
https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua/1.4.356.27

kieranhaycock, 2019-08-29:
Thanks for the quick response. 

The Nuget and a lot of the sample code have class names that are the same. 

I'm using the "DataAccess Server" project as a template and wanted to remove the reference to Opc.Ua.Core, Server & Configuration in favour of the Nuget package. When I did this I got build errors because MonitoredNode was missing a load of functions and it was marked as deprecated. It looked as though it should be using MonitoredNode2 so I changed to that but as above I then have to use CustomNodeManager2.

It looks as though MonitoredNode2 was copied in the samples to MonitoredNode looking at the comment summary above the constructor. That is why it seemed as though the Nuget was out of date. I thought it had been replaced but looking deeper into the code it starts to be a little less complicated.

Should I use CustomNodeManager2? Does it provide most of the same functionality as QuickstartNodeManager in the samples?

AlinMoldovean, 2019-08-29:
@kieranhaycock ,

You can extend CustomNodeManager2 for your node manager implementation.

closed 2019-09-19 by AlinMoldovean