OPCFoundation/UA-.NETStandard #787: Import output from OPC UA-ModelCompiler into address space

james-sys, 2019-08-31:
So I have successfully followed Stefan Profanter’s guide and created the model files using the UA-ModelCompiler. But how do I use these in .NET? I have got as far as the the CreateAddressSpace in the NodeManager but, I am struggling to find how to use the NodeSet2, uanodes, .xsd and other files produced by the ModelCompiler. Any help is appreciated.

MD-V, 2019-09-01:
@james-sys ,

please check the DataTypes example. Especially [the LoadPredefinedNodes](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/Workshop/DataTypes/Server/DataTypesNodeManager.cs#L164) method.

james-sys, 2019-09-01:
@MD-V  Thanks. I cannot find any examples that utilise the NodeSet2.xml to initialize the address space.

MD-V, 2019-09-01:
@james-sys You dont need the NodeSet file if you have a uanodes file.

james-sys, 2019-09-01:
@MD-V Thanks for your help

closed 2019-09-01 by james-sys