OPCFoundation/UA-.NETStandard #789: Not fully .Net core compatible

damikun, 2019-09-02:
When i try to run or work with examples on OSX using .net core console sample which is included, im not able to do that because it has dependencies to opcua.core library that is not compatible with osx and .net core...

MD-V, 2019-09-02:
@damikun 

Please be more specific. Which libary is not compatible and which .NetCore version are you using? 
What is the error message?

As the libary is .Net Standard 2.0 compatible it should also be .Net Core 2.0 compatible.

damikun, 2019-09-03:
Hi Yes sorry

First I'm running it in VS mac osx... Im using .net core 2+ usually 2.1.. but now I'm just trying to run OPC Foundation OPCUA console client example.. 

The first problem is after opening solution.. The packages / project dependence cannot get restored (first error) i cannot do restor...  if i deleted this dependence and try to add them manually it is not possible. => The console app is in .net core but OPCUA.Core is in NetStandard thats error..

mregen, 2019-10-01:
Hi @damikun, your approach to open the solutions in VS Code has not been tested and is therefore not supported. As you can see, the UI cannot distinguish the librararies which build against multiple targets, e.g. net46, netstandard13, netstandard 20.
Also the .Net 4.6 sample apps with UI are not supported on other OS than Windows.
What is supported is to open a .net core .csproj in vscode, it should then be able to build and run the libs as netstandard20 etc. Makes sense?

closed 2020-01-10 by mregen