OPCFoundation/UA-.NETStandard #790: WriteValue Structs with EncodeableObjects

remusjones, 2019-09-03:
I'm having an issue writing a "complex structure".

Here's the EncodeableObject I am using: 

    public class Cyl_Status : EncodeableObject
    {
        public bool ADVD_Reed { get; set; }
        public bool RETD_Reed { get; set; }
        public bool Advance_Sol { get; set; }
        public bool Retract_Sol { get; set; }
        public bool Man_Advance_Vis { get; set; }
        public bool Man_Retract_Vis { get; set; }
        public override ExpandedNodeId TypeId => ExpandedNodeId.Parse("nsu=urn:OMRON:NxOpcUaServer:FactoryAutomation;i=5002");

        public override ExpandedNodeId BinaryEncodingId => ExpandedNodeId.Parse("nsu=urn:OMRON:NxOpcUaServer:FactoryAutomation;i=76");

        public override ExpandedNodeId XmlEncodingId => NodeId.Null;
        public Cyl_Status() { }
        public Cyl_Status(bool advd, bool retd, bool advance, bool retract, bool manAdvance, bool manRetract)
        {
            ADVD_Reed = advd;
            RETD_Reed = retd;
            Advance_Sol = advance;
            Retract_Sol = retract;
            Man_Advance_Vis = manAdvance;
            Man_Retract_Vis = manRetract;
        }
        public override void Encode(IEncoder encoder)
        {
            encoder.WriteBoolean("ADVD_Reed", this.ADVD_Reed);
            encoder.WriteBoolean("RETD_Reed", this.RETD_Reed);
            encoder.WriteBoolean("Advance_Sol", this.Advance_Sol);
            encoder.WriteBoolean("Retract_Sol", this.Retract_Sol);
            encoder.WriteBoolean("Man_Advance_Vis", this.Man_Advance_Vis);
            encoder.WriteBoolean("Man_Retract_Vis", this.Man_Retract_Vis);
        }

        public override void Decode(IDecoder decoder)
        {
            ADVD_Reed = decoder.ReadBoolean("ADVD_Reed");
            RETD_Reed = decoder.ReadBoolean("RETD_Reed");
            Advance_Sol = decoder.ReadBoolean("Advance_Sol");
            Retract_Sol = decoder.ReadBoolean("Retract_Sol");
            Man_Advance_Vis = decoder.ReadBoolean("Man_Advance_Vis");
            Man_Retract_Vis = decoder.ReadBoolean("Man_Retract_Vis");
        }

        public override string ToString() => $"{{ ADVD_Reed={this.ADVD_Reed}; RETD_Reed={this.RETD_Reed}; Advance_Sol={this.Advance_Sol}; Retract_Sol={this.Retract_Sol}; Man_Advance_Vis={this.Man_Advance_Vis}; Man_Retract_Vis={this.Man_Retract_Vis}; }}";

        public bool _ADVD_Reed { get => ADVD_Reed; set => ADVD_Reed = value; }
        public bool _RETD_Reed { get => RETD_Reed; set => RETD_Reed = value; }
        public bool _Advance_Sol { get => Advance_Sol; set => Advance_Sol = value; }
        public bool _Retract_Sol { get => Retract_Sol; set => Retract_Sol = value; }
        public bool _Man_Advance_Vis { get => Man_Advance_Vis; set => Man_Advance_Vis = value; }
        public bool _Man_Retract_Vis { get => Man_Retract_Vis; set => Man_Retract_Vis = value; }

I have this in the init section:
`EncodeableFactory.GlobalFactory.AddEncodeableType(typeof(Containers.Cyl_Status));`

I am retrieving the data from the subscription event like this:
` var result = notification.Value.GetValue(typeof(Containers.Cyl_Status));`
 ` cylStatus = result as Containers.Cyl_Status;`

I modify the data and attempt to send it back: 

`WriteValue wv = new WriteValue();`
`wv.NodeId = new NodeId(nodeid);`
`wv.AttributeId = Attributes.Value;`
`var t = new ExtensionObject(CylStatus.TypeId);`
`t.Body = CylStatus;`
`wv.Value = new DataValue(new Variant(t));`

In this case I get a "BadInvalidArgument" Code.
I've tried a bunch of different ways to parse the value to the plc, maybe I am doing it wrong.
I've been stuck on this issue for a week now, so any assistance would be greatly appreciated. 

I can modify the struct inside UAExpert, so it has Read/Writable.

awcullen, 2019-09-03:
For testing structures, I am using the UaCPPServer from Unified Automation. There is a simple structure called Vector.  Here is an example of reading and writing.

To run this code, you will need two additional xml files to the directory. (see following posts)

```
using Opc.Ua;
using Opc.Ua.Client;
using Opc.Ua.Configuration;
using System;
using System.Collections.Generic;
using System.Threading;
using System.Threading.Tasks;

namespace ConsoleClient
{
    public class Program
    {
        private static void Main(string[] args)
        {
            try
            {
                Task.Run(() => RunAsync()).GetAwaiter().GetResult();
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.Message);
            }
            Console.WriteLine("Press any key to close the program...");
            Console.ReadKey(true);
        }

        private static async Task RunAsync(CancellationToken token = default(CancellationToken))
        {
            Console.WriteLine("Create an Application Instance.");

            var application = new ApplicationInstance
            {
                ApplicationName = "Console Client",
                ApplicationType = ApplicationType.Client,
                ConfigSectionName = "ConsoleClient"
            };

            // load the application configuration.
            ApplicationConfiguration config = await application.LoadApplicationConfiguration(false);

            // check the application certificate. If cert not found, create a self-signed one.
            bool haveAppCertificate = await application.CheckApplicationInstanceCertificate(false, 0);
            if (!haveAppCertificate)
            {
                throw new Exception("Application instance certificate invalid!");
            }
            // add handler to validate the server cert (if it is not already in the trusted list).
            config.CertificateValidator.CertificateValidation += new CertificateValidationEventHandler(OnCertificateValidation);

            // load first endpoint from file
            ConfiguredEndpointCollection endpoints = config.LoadCachedEndpoints(true);
            if (endpoints.Count == 0)
            {
                throw new Exception("Endpoints file invalid!");
            }
            var selectedEndpoint = endpoints[0];
            selectedEndpoint.UpdateFromServer();

            Console.WriteLine("Selected Endpoint:");
            Console.WriteLine("  Server Uri: {0}", selectedEndpoint.Description.Server.ApplicationUri);
            Console.WriteLine("  Server Certificate: byte[{0}]", selectedEndpoint.Description.ServerCertificate.Length);
            Console.WriteLine("  Security Policy: {0}", selectedEndpoint.Description.SecurityPolicyUri.Split('#', 2)[1]);
            Console.WriteLine("  Security Mode: {0}", selectedEndpoint.Description.SecurityMode);

            Console.WriteLine("Create a session with OPC UA server.");
[... 80 lines cut ...]

        public override ExpandedNodeId TypeId => ExpandedNodeId.Parse("nsu=http://www.unifiedautomation.com/DemoServer/;i=3002");

        public override ExpandedNodeId BinaryEncodingId => ExpandedNodeId.Parse("nsu=http://www.unifiedautomation.com/DemoServer/;i=5054");

        public override ExpandedNodeId XmlEncodingId => NodeId.Null;

        public override void Encode(IEncoder encoder)
        {
            encoder.WriteDouble("X", this.X);
            encoder.WriteDouble("Y", this.Y);
            encoder.WriteDouble("Z", this.Z);
        }

        public override void Decode(IDecoder decoder)
        {
            this.X = decoder.ReadDouble("X");
            this.Y = decoder.ReadDouble("Y");
            this.Z = decoder.ReadDouble("Z");
        }

        public override string ToString() => $"{{ X={this.X:f2}; Y={this.Y:f2}; Z={this.Z:f2}; }}";
    }
}
```

awcullen, 2019-09-03:
ConsoleClient.Config.xml

```
<?xml version="1.0" encoding="utf-8"?>
<ApplicationConfiguration
  xmlns:ua="http://opcfoundation.org/UA/2008/02/Types.xsd"
  xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd"
  schemaLocation="./Schema/ApplicationConfiguration.xsd"
>
  <ApplicationName>Console Client</ApplicationName>
  <ApplicationUri>urn:localhost:beeond.net:ConsoleClient</ApplicationUri>
  <ProductUri>http://beeond.net/ConsoleClient</ProductUri>
  <ApplicationType>Client_1</ApplicationType>

  <SecurityConfiguration>

    <!-- Where the application instance certificate is stored -->
    <ApplicationCertificate>
      <StoreType>Directory</StoreType>
      <StorePath>%ProgramData%/OPC Foundation/pki/own</StorePath>
      <SubjectName>CN=Console Client, C=US, S=NewYork, O=OPC Foundation, DC=localhost</SubjectName>
    </ApplicationCertificate>

    <!-- Where the issuer certificate are stored (certificate authorities) -->
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%ProgramData%/OPC Foundation/pki/issuer</StorePath>
    </TrustedIssuerCertificates>

    <!-- Where the trusted certificates are stored -->
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%ProgramData%/OPC Foundation/pki/trusted</StorePath>
    </TrustedPeerCertificates>

    <!-- The directory used to store invalid certficates for later review by the administrator. -->
    <RejectedCertificateStore>
      <StoreType>Directory</StoreType>
      <StorePath>%ProgramData%/OPC Foundation/pki/rejected</StorePath>
    </RejectedCertificateStore>

    <!-- WARNING: The following setting (to automatically accept untrusted certificates) should be used
    for easy debugging purposes ONLY and turned off for production deployments! -->
    <AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>

  </SecurityConfiguration>

  <TransportConfigurations/>

  <TransportQuotas>
    <OperationTimeout>600000</OperationTimeout>
    <MaxStringLength>1048576</MaxStringLength>
    <MaxByteStringLength>4194304</MaxByteStringLength>
    <MaxArrayLength>65535</MaxArrayLength>
    <MaxMessageSize>4194304</MaxMessageSize>
    <MaxBufferSize>65535</MaxBufferSize>
    <ChannelLifetime>300000</ChannelLifetime>
    <SecurityTokenLifetime>3600000</SecurityTokenLifetime>
  </TransportQuotas>

  <ServerConfiguration/>

  <ClientConfiguration>
    
    <!-- The default timeout for new sessions -->
    <DefaultSessionTimeout>600000</DefaultSessionTimeout>

    <!-- The well-known URLs for the local discovery servers
         URLs are tested in the order they appear in this list. -->
    <WellKnownDiscoveryUrls>
      <ua:String>opc.tcp://{0}:4840/UADiscovery</ua:String>
    </WellKnownDiscoveryUrls>

    <!-- EndpointDescriptions for system wide discovery servers -->
    <DiscoveryServers></DiscoveryServers>

    <!-- A file that stores the cached endpoints. -->
    <EndpointCacheFilePath>ConsoleClient.Endpoints.xml</EndpointCacheFilePath>

    <!-- The minimum subscription lifetime.
      This ensures subscriptions are not set to expire too quickly. The requesed lifetime count
      and keep alive count are calculated using this value and the request publishing interval -->
    <MinSubscriptionLifetime>10000</MinSubscriptionLifetime>  
  </ClientConfiguration>

  <DiscoveryServerConfiguration/>

  <XmlElementCollection/>
  
  <TraceConfiguration>
    <OutputFilePath>%ProgramData%/Logs/ConsoleClient.log.txt</OutputFilePath>
    <DeleteOnLoad>true</DeleteOnLoad>
    <!-- Show Only Errors -->
    <!-- <TraceMasks>1</TraceMasks> -->
    <!-- Show Only Security and Errors -->
    <!-- <TraceMasks>513</TraceMasks> -->
    <!-- Show Only Security, Errors and Trace -->
    <!-- <TraceMasks>515</TraceMasks> -->
    <!-- Show Only Security, COM Calls, Errors and Trace -->
    <!-- <TraceMasks>771</TraceMasks> -->
    <!-- Show Only Security, Service Calls, Errors and Trace -->
    <!-- <TraceMasks>523</TraceMasks> -->
    <!-- Show Only Security, ServiceResultExceptions, Errors and Trace -->
    <TraceMasks>519</TraceMasks>
  </TraceConfiguration>

  <!-- Disables the hi-res clock if the QueryPerformanceCounter does work on a particular machine. -->
  <DisableHiResClock>true</DisableHiResClock>

</ApplicationConfiguration>
```

awcullen, 2019-09-03:
ConsoleClient.Endpoints.xml
```
<?xml version="1.0" encoding="utf-8"?>
<ua:ConfiguredEndpointCollection
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd"
  xmlns:ua="http://opcfoundation.org/UA/SDK/Configuration.xsd">
  <ua:KnownHosts />
  <ua:Endpoints>
    <ua:ConfiguredEndpoint>
      <ua:Endpoint>
        <EndpointUrl>opc.tcp://localhost:48010</EndpointUrl>
        <SecurityMode>SignAndEncrypt_3</SecurityMode>
        <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256</SecurityPolicyUri>
        <UserIdentityTokens>
          <UserTokenPolicy>
            <TokenType>Anonymous_0</TokenType>
          </UserTokenPolicy>
        </UserIdentityTokens>
        <TransportProfileUri>http://opcfoundation.org/UA/profiles/transport/wsxmlorbinary</TransportProfileUri>
      </ua:Endpoint>
      <ua:UpdateBeforeConnect>true</ua:UpdateBeforeConnect>
      <ua:SelectedUserTokenPolicy>0</ua:SelectedUserTokenPolicy>
    </ua:ConfiguredEndpoint>
  </ua:Endpoints>
</ua:ConfiguredEndpointCollection>
```

awcullen, 2019-09-03:
Another thing to check is the nodeids.  The TypeId is found by searching for the NodeId of the DataType 'Vector".  The BinaryEncodingId is not usually found in the namespace, but you can find it with UaExpert by right clicking the "HasEncoding" reference named "Default Binary" and choosing "Follow Reference".

[image]

remusjones, 2019-09-04:
> Another thing to check is the nodeids. The TypeId is found by searching for the NodeId of the DataType 'Vector". The BinaryEncodingId is not usually found in the namespace, but you can find it with [...]

You're a life saver, it turns out I had my Binary and TypeID wrong in the struct. 

[image]
[image]

closed 2019-09-04 by remusjones

Neil336, 2023-06-16:
I have a similar issue: I also have a custom datatype, implemented in the same way as both of the examples above. However, passing the value directly yields a BadTypeMismatch and thus I tried to make use of the XmlEncoder. But this was also not successful.
```
EncodeableFactory.GlobalFactory.AddEncodeableType(typeof(CustomType));
IEncodeable encodable = new CustomType();
var val = EncodeXml(new CustomType());

WriteValue nodeToWrite = new WriteValue()
{
    NodeId = "ns=4;s=...", // here I have the actual NodeId
    AttributeId = Attributes.Value,
    Value = new DataValue(new Variant(val))
};
```

I had to modify the EncodeXml function a bit to avoid Stack Empty error:
```
public static ExtensionObject EncodeXml(IEncodeable encodable)
{
  // create encoder.
  XmlEncoder encoder = new XmlEncoder(typeof(RunInType), null, ServiceMessageContext.GlobalContext);
  
  // write body.
  encoder.WriteExtensionObjectBody(encodable);
  
  // create document from encoder.
  XmlDocument document = new XmlDocument();
  using (var sreader = new StringReader(encoder.Close()))
  using (var reader = XmlReader.Create(sreader, DefaultXmlReaderSettings()))
  {
      document.XmlResolver = null;
      document.Load(reader);
  }
  XmlElement body = document.DocumentElement;
  
  // return root element.
  return new ExtensionObject(encodable.XmlEncodingId, body);
  }
  
  /// <summary>
  /// Returns a XmlReaderSetting with safe defaults.
  /// DtdProcessing Prohibited, XmlResolver disabled and
  /// ConformanceLevel Document.
  /// </summary>
public static XmlReaderSettings DefaultXmlReaderSettings()
{
  return new XmlReaderSettings()
  {
      DtdProcessing = DtdProcessing.Prohibit,
      XmlResolver = null,
      ConformanceLevel = ConformanceLevel.Document
  };
}
```
Could you maybe help me?