OPCFoundation/UA-.NETStandard #792: Get Access to RegisterNodes in Server Implementation

tschombe, 2019-09-05:
Hi,

I habe a server implementation, where my NodeManager derives from
CustomNodeManager2

To optimize read / write operations, I want to override the RegisterNodes() method
to provide my own handles to the client.

I cannot override RegisterNodes() in my implementation since this method only
exist in MasterNodeManager.

How should I implement this?
Should I derive from MasterNodeManager instead of CustomNodeManager2?

whats the recommended approach here?

AlinMoldovean, 2019-09-05:
@tschombe ,

Yes, the implementation for _RegisterNodes_ and _UnregisterNodes_ should be implemented at `MasterNodeManager `level.
Or in `SessionServerBase`

tschombe, 2019-09-06:
Thanks, that helped me, I did not know that SessionServerBase also supports it, I can override it there.

closed 2019-09-06 by tschombe