OPCFoundation/UA-.NETStandard #796: Session.cs - ReadValues Method delivers Old Values

mz-zmsoft, 2019-09-17:
The Session Method "ReadValues" delivers old values. This is not the behavior I expect. Method calls the Read Method an passing Int32.MaxValue for Parameter "maxAge".  Tested on SIMATIC NET OPC UA Server.

```
  ResponseHeader responseHeader = Read(
                null,
                Int32.MaxValue,
                TimestampsToReturn.Both,
                valuesToRead,
                out results,
                out diagnosticInfos);
```

MD-V, 2019-09-17:
@mz-zmsoft ,

unfortunately the information you are giving is not sufficient. There are many people that are using this functionality without problems.

Can you please attach a picture of a connected UAExpert that shows the current value and your OPC UA client showing the incorrect value?

mz-zmsoft, 2019-09-17:
Hey, 

what happens is following... SIMATIC NET OPC UA Server is hosting a Node (DataType = STRING). 

- On the PLC I set the Variable Value to "TEST". 
- I'm starting a very simple Client App I created for testing purposes of this issue. It creates a session and reads the Variable cyclic. That's it.

```
// Snippet 1
 var session = await Session.Create(config, endpoint, false, "OPC UA Console Client", 60000, new UserIdentity(new AnonymousIdentityToken()), null);

while (true) 
{
    session.ReadValues(new List<NodeId>() { nodeId }, new List<Type>() { typeof(string) }, out var values, out var errors);

    Thread.Sleep(3000);

    System.Console.WriteLine($"Value: {values[0]}");

}

```
- When Starting the App the Output for the Value is always "Value: TEST", as expected.
- While the App is running I modify the Value on PLC to "TEST2". 
- The running App still writes "Value: TEST" instead of "Value: TEST2" in Output.
- I quit the App now, and restart it completely but still got the Output "Value: TEST". Seems like it's reading old values from cache on OPC UA Server!?

What I did now, is modifying my code to following:

```
// Snippet 2
while (true) 
{
    session.ReadValues(new List<NodeId>() { nodeId }, new List<Type>() { typeof(string) }, out var values, out var errors);

    var dv = session.ReadValue(nodeId);

    System.Console.WriteLine($"Value: {values[0]}");
    
    System.Console.WriteLine($"Single Read Value: {dv.Value}");

    Thread.Sleep(3000);
}
```

Now starting with my test at the beginning.

- PLC Value is 'TEST'
- Output is : 
```
Value: TEST
Single Read Value: Test
```
- Updating the Value on PLC to 'TEST2'
- Output is : 
```
Value: TEST
Single Read Value: TEST2
```

Other scenario that works, when using Code Snippet 1 is following:

- Run the App
- Update the Plc Value --> still got the wrong ('TEST') Value in output.
- Navigating to the Node in UA Expert an doing a refresh on target Node.
- The App immediatly displays the new Value

HOPE THIS HELPS!!!

mz-zmsoft, 2019-09-17:
From Sesson.cs  - See the difference?

 ```
  public DataValue ReadValue(NodeId nodeId)
        {
         ...
            ResponseHeader responseHeader = Read(
                null,
                0,  // <-- maxAge = 0
                TimestampsToReturn.Both,
                itemsToRead,
                out values,
                out diagnosticInfos);
         ...
        }

 public void ReadValues(
            IList<NodeId> variableIds,
            IList<Type> expectedTypes,
            out List<object> values,
            out List<ServiceResult> errors)
        {
        ...
            ResponseHeader responseHeader = Read(
                null,
                Int32.MaxValue, // <-- maxAge = Int32.MaxValue
                TimestampsToReturn.Both,
                valuesToRead,
                out results,
                out diagnosticInfos);
        ...            
        }
```

closed 2019-09-19 by AlinMoldovean

fivage, 2022-08-18:
The doc of OPCFoundation is maybe confusing:
_If maxAge is set to the max Int32 value or greater, the Server shall attempt to get a cached value._

If you specify Int32.Max, then you will do a physical device read once, and then keep on getting the same value for the next 23 days, which is probably never your goal...
Or am I misinterpreting this...?

commit b1c0293f (2019-09-19): Fixes #796