OPCFoundation/UA-.NETStandard #803: make Part 100: Device Information Model specification avialable for custom Information Model [QUESTION]

STUBA-rupr, 2019-09-23:
It seems that _Part 100: Device Information Model_ specification is part of [OPC Unified Architecture Specification](https://opcfoundation.org/developer-tools/specifications-unified-architecture).
However it is not part of the Opc.Ua.Core stack.
Also repositories [UA-Nodeset](https://github.com/OPCFoundation/UA-Nodeset), or [UA-ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler) does not contain DI objects in _Scheme_ folder.
Does it mean that I need to add DI files into _UA-.NETStandard/Stack/Opc.Ua.Core/Stack/Generated/_ folder manually? If it is so, I need to take them from [UA-Nodeset](https://github.com/OPCFoundation/UA-Nodeset), or [UA-ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler)? 
Currently I have issues during Core compiling with both of them.
Or how can I use my [information model](https://github.com/STUBA-rupr/DI-InformationModel/blob/master/Design/modeldesignscale.xml) (created with ModelCompiler) based on DI in OPC UA server?

MD-V, 2019-09-23:
@STUBA-rupr,

Currently this repository and the other repositories are not in sync (starts at the core OPC UA definitions).
I think you will have to wait until  #791 is resolved.

STUBA-rupr, 2019-09-24:
Hi @MD-V, the information from issue #791, that 1.04 branch as source for Nodeset in NETStandard was useful for me. While the issue is being resolved I can proceed.

closed 2019-09-24 by STUBA-rupr