OPCFoundation/UA-.NETStandard #806: More than one  DA addresse

mary7091, 2019-09-25:
Hi, 
Can this wrapper converts more than one DA address to UA addresses at the same time ???

awcullen, 2019-09-25:
Yes, one COMServerWrapper can wrap multiple DA servers.  Add the server's info to the .config.xml like I have below:

```
  <Extensions>
    <ua:XmlElement>
      <ComWrapperServerConfiguration xmlns:i="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://opcfoundation.org/UA/SDK/COMInterop">
        <WrappedServers>
          <ComClientConfiguration i:type="ComDaClientConfiguration">
            <ServerUrl>opc.com://localhost/Matrikon.OPC.Simulation</ServerUrl>
            <ServerName>MA</ServerName>
            <MaxReconnectWait>10000</MaxReconnectWait>
            <SeperatorChars></SeperatorChars>
            <AvailableLocales>
              <ua:String>en-US</ua:String>
            </AvailableLocales>
            <BrowseToNotSupported>false</BrowseToNotSupported>
          </ComClientConfiguration>
          <ComClientConfiguration i:type="ComDaClientConfiguration">
            <ServerUrl>opc.com://localhost/OPC.SimaticNET</ServerUrl>
            <ServerName>S7</ServerName>
            <MaxReconnectWait>10000</MaxReconnectWait>
            <SeperatorChars></SeperatorChars>
            <AvailableLocales>
              <ua:String>en-US</ua:String>
            </AvailableLocales>
            <BrowseToNotSupported>false</BrowseToNotSupported>
          </ComClientConfiguration>
          <ComClientConfiguration i:type="ComDaClientConfiguration">
            <ServerUrl>opc.com://localhost/Softing.OPCToolboxDemo_ServerDA.1</ServerUrl>
            <ServerName>SO</ServerName>
            <MaxReconnectWait>10000</MaxReconnectWait>
            <SeperatorChars></SeperatorChars>
            <AvailableLocales>
              <ua:String>en-US</ua:String>
              <ua:String>de-DE</ua:String>
              <ua:String>ja-JP</ua:String>
            </AvailableLocales>
            <BrowseToNotSupported>false</BrowseToNotSupported>
          </ComClientConfiguration>
        </WrappedServers>
      </ComWrapperServerConfiguration>
    </ua:XmlElement>
  </Extensions>
```

mregen, 2019-10-01:
Hi @mary7091, please close the issue if you feel your issue has been resolved. Thanks.

closed 2019-10-01 by mary7091