OPCFoundation/UA-.NETStandard #822: CertificateValidator doesn't call validation event for expired certificates

berndbr, 2019-10-14:
Having a server that uses a self-signed certificate the method InternalValidate of CertificateValidator always throws a ServiceResultException with StatusCode BadSecurityChecksFailed because the servers certificate is untrusted (not contained in the list of trusted issuers).
See this line of code:

```csharp
foreach (X509ChainStatus status in element.ChainElementStatus)
{
    ServiceResult result = CheckChainStatus(status, target, issuer, (ii != 0));

    if (ServiceResult.IsBad(result))
    {
        // check untrusted certificates.
        if (trustedCertificate == null)
        {
            throw new ServiceResultException(StatusCodes.BadSecurityChecksFailed);
        }

        throw new ServiceResultException(result);
    }
    chainStatusChecked = true;
}
```

In this case the client isn't able to accept the certificate as throwing the exception with BadSecurityChecksFailed the certificate gets rejected in the exception handler containedin the Validate method aftwerwards.

Is there a good reason why the above isn't implemented like this:

```csharp
foreach (X509ChainStatus status in element.ChainElementStatus)
{
    ServiceResult result = CheckChainStatus(status, target, issuer, (ii != 0));

    if (ServiceResult.IsBad(result))
    {
        // check untrusted certificates.
        // if (trustedCertificate == null)
        // {
        //    throw new ServiceResultException(StatusCodes.BadSecurityChecksFailed);
        // }

        throw new ServiceResultException(result);
    }
    chainStatusChecked = true;
}
```
With this the reason StatusCode BadCertificateIssuerTimeInvalid will be thrown which aftwards is allowed to be handled in the CertificateValidation event.

I'm using version 1.4.357.28 of the SDK.

AlinMoldovean, 2019-10-14:
Hi @berndbr ,

The validation event is triggered only for validation errors that are considered suppressible. i.e. the certificate is valid, but it is not yet trusted, or the certificate is trusted but is has a validation error that the user should be informed about.

In your case the certificate is both expired and not trusted. In this case an offline validation is required (by checking the rejected store.)

berndbr, 2019-10-14:
Hi @AlinMoldovean ,

thanks for the fast reply.

Anyway I would expect to be able to handle the validation error. It's then in the decision of the use case whether such certificates should be accepted or not (preferrable they should not be accepted of course). I currently have such a use case where it is necessary to accept such certificates.

Or at least - shouldn't the user be informed what's wrong with the certificate? By always overwriting the validations result StatusCode with BadSecurityChecksFailed the user won't get a hint about what's wrong with the certificate, just that somethings wrong with it.

PaulHunkar, 2019-10-17:
From a security point of view, unless a certificate is trusted no
additional information shall be provided.  That does not mean that the
server should not be recording the additional information about what was
actually wrong in a log file or in an event for an audit system. If the
additional information were provided, the  additional information might be
leverage to attack the system.  For all error the same information shall be
reported preferably with the same timing.

Paul

marcschier, 2019-10-22:
@AlinMoldovean, this is first and foremost an API issue for me.  _If_ there is a validation override, expired certificates should be handed to the override just like any other certificate validation errors.  I see no difference between expiration and for example a bad subject name.  If a client (and user) wants to trust a certificate that is expired or more important - not yet valid, it should be able to, without requiring someone to move the certificate into another folder.

commit dc9fa85a (2019-11-06): Fixes #822 - Certificate Validation fix. (#843)

* Remove BadCertificateUriInvalid from list of certificate validation errors that may be suppressed.
closed 2019-11-06 by AlinMoldovean