OPCFoundation/UA-.NETStandard #826: Creating nodes as data comes in

puppetSpace, 2019-10-17:
The application i'm writing is receiving data in the form of JSON. 
The application is also hosting a StandardServer
From the JSON data I get which nodes should be created and what value they have.
I check if the nodes already exist. If they do not exist, I create them and then write data to them.
This is all done on the server, there is no OPC client

But I just can't seemed to find out how to create the nodes once the server is running. There are examples enough on how to load predefined nodes, but not on how to add nodes when the server is started.

Do I have to access CustomNodeManager.AddPredefinedNode somehow?

AlinMoldovean, 2019-10-17:
Hi @puppetSpace ,

The nodes can be created during the run-time in the same way as they are created in CreateAddressSpace() method from server samples.
The steps are to create the node, set all required attributes and call AddPredefinedNode()

puppetSpace, 2019-10-18:
It isn't working. The created nodes aren't showing up in the client (even after a reconnect)
My test [https://github.com/puppetSpace/Labs/tree/master/ServerTest](https://github.com/puppetSpace/Labs/tree/master/ServerTest).

In Program.cs, I start the server and then add the nodes:
https://github.com/puppetSpace/Labs/blob/27651eebe16ae561c91795fdc4e6c3707e329adb/ServerTest/Program.cs#L26-L31

When I move this piece of code to MyNodeManager.CreateAddressSpace, the nodes are available in the client.

What am I missing?

puppetSpace, 2019-10-18:
hmm, it seems that a Root node should already be defined. Then you can add nodes to it

AlinMoldovean, 2019-10-18:
@puppetSpace ,

The problem is that the _rootNode_ of your structure does not have references, so it is not reachable through browsing.

closed 2019-10-28 by AlinMoldovean