OPCFoundation/UA-.NETStandard #827: Automatically register Type with BinaryEncodingId

wim07101993, 2019-10-18:
Hi,

I have build a wrapper around the EncodableObject that does the encoding/decoding automatically. But now I also want to automate the fetching of the TypeId. Is this possible? I know I can get the type Id with 
```
var value = client.Session.ReadValue(NodeId.Parse("ns=4;s=MyVector"));
if (value.Value is ExtensionObject extensionObject)
{
    var typeId = extensionObject.TypeId;
    ....
}
```
But can I than use that type to register a class with the  `EncodeableFactory`? As far as i know the only way to register a type is using `EncodeableFactory.GlobalFactory.AddEncodeableType(type)`.

AlinMoldovean, 2019-10-28:
Hi @wim07101993 ,

You need to call `ClientSession.Factory.AddEncodeableType()` specifying the _typeId_  and and the type that implements the `Encode()` and `Decode()` methods for that `TypeId`.

wim07101993, 2019-10-28:
Hi,

I would like to let the program determine the TypeId by discovering it itself. That way I would not need to go and search for it each time it changes. Is it also possible to determine the TypeId at runtime?

AlinMoldovean, 2019-10-29:
@wim07101993 ,

Only the OPC UA predefined types (the encodable types from _Core_ library) are automatically added to EncodeableFactory

AlinMoldovean, 2019-11-21:
Hi @wim07101993 ,

An implementation for loading custom complex types from server was added in #849

closed 2019-11-22 by AlinMoldovean