OPCFoundation/UA-.NETStandard #832: Possibility for StackOverflowException in TcpMessageSocket

BlueManiac, 2019-10-23:
We have identified a potential recursion in TcpMessageSocket.cs.

One of our applications has suffered a StackOverflowException that we have traced to recursive loop of ReadNextBlock() -> OnReadComplete() -> DoReadComplete() -> ReadNextBlock() and so on.

This seems to occur whenever there are many open subscriptions and/or whenever a lot of messages are received in a short amount of time.

We were only able to reproduce this when running without debug.

[image]

[image]

mregen, 2019-10-23:
Thanks @BlueManiac, indeed there can be a stack overflow if ReceiveAsync decides to always return synchronous (optimization since .net core 2). A fix is needed.

closed 2019-10-30 by AlinMoldovean

Wadjet-Informatics, 2019-11-11:
@AlinMoldovean Will this be making an official release in the near future?

AlinMoldovean, 2019-11-11:
@Wadjet-Informatics ,

Yes. The fix is already on the master branch and the next Nuget Package release will include it.

Wadjet-Informatics, 2019-11-11:
@AlinMoldovean Thank you! We have tried the fix but encountered some performance issues. Could you take a look at my PR and see if you think it makes sense?

commit f015ece8 (2019-10-30): Fixes #832 - Avoid deep recursion in TcpMessageSocket.ReadNextBlock() (#836)

* Fixes #832 - Avoid deep recursion in TcpMessageSocket.ReadNextBlock()