OPCFoundation/UA-.NETStandard #837: Integration of OPC UA .net standard build to Apache NiFi

vraghu1983, 2019-10-30:
Hi - wanted to understand how does this new .Net standard integrate with Apache NiFi tool.

I have already used the Java legacy project to build and extract .nar files to be integrated with NiFi. 
Wanted to understand if a similar thing is possible using this source code as it is the only one actively maintained for OPC UA data extraction

MD-V, 2019-10-31:
Hello @vraghu1983,

NAR packages are based on Java libaries. Is this correct? As this is a C# based libary you would need a wrapper around the libary to call it from Java (Could be done by using [jni4net](https://github.com/jni4net/jni4net/))

vraghu1983, 2019-10-31:
Oh yes. This is java based.
Ok if there are articles on this, will be helpful for me

closed 2020-11-06 by AlinMoldovean