OPCFoundation/UA-.NETStandard #838: Complex Type Client extension library prototype for testing

mregen, 2019-10-31:
In the Prototyping branch: https://github.com/OPCFoundation/UA-.NETStandard/tree/prototyping_complextypeclient there is a prototype to extend the type system of a client with custom enumerations and structure types based on a server dictionary or DataTypeDefinition using emitted classes at runtime. Support includes V1.03 dictionaries and V1.04 DataTypeDefinitions (--> known issues).

In the prototyping branch there is the UA Sample Client, the Quickstart Data Type client and the NetCoreComplexClient using the library and can be used for testing.

Finding enough servers with custom types is an issue, so if you find servers with unsupported custom types please provide feedback.

Known issues:
- UA Sample Client hits plenty of UI issues when writing custom types.
- Only V1.03 binary dictionaries are supported (no XML).
- V1.03 dictionary structure definitions are only supported if they can be mapped to a V1.04 DataTypeDefinition.
- OptionSet is only partially implemented.
- Opaque Types are ignored

mregen, 2019-10-31:
Please vote here if the complex type library should be included OPC Foundation library package.

awcullen, 2019-10-31:
Works well!
I don't know if the order of properties returned from GetProperties() is guaranteed.  There is a good discussion here about some workarounds.
https://stackoverflow.com/questions/9062235/get-properties-in-order-of-declaration-using-reflection

mregen, 2019-11-04:
Hi @awcullen, good point, I will check the order of the properties. They have the order in the Datamember attribute, it can be used for sorting... Cheers, Martin

MaartenBE, 2019-11-14:
I think this is a very important feature and is definitely a very nice addition.

closed 2019-11-20 by mregen