OPCFoundation/UA-.NETStandard #842: Subscription-requests for non existing nodes slows down server

DanielZ87, 2019-11-04:
Hi

The other day I had the problem that my OPC UA server was overloaded. After some investigation, I found out, that a client was always trying to subscribe with wrong NodeIds.

Do you have an idea how to deal with such clients? 

I know it is not really the fault of the server in first place, but there should be a way to prevent such a tear down.

thx for any suggestion

AlinMoldovean, 2019-11-05:
Hi @DanielZ87 ,

The server should return BadNodeIdUnknown for invalid monitored nodes.

You can avoid this if you expose only endpoints with security and have a predefined list of trusted clients that can connect.

DanielZ87, 2019-11-05:
> Hi @DanielZ87 ,
> 
> The server should return BadNodeIdUnknown for invalid monitored nodes.
> 
> You can avoid this if you expose only endpoints with security and have a predefined list of trusted clients that can connect.

Hi @AlinMoldovean 

Thx for the reply. To your answer I would say yes and no. It does reduce the chances but not solve the problem in its origin. I mean this problem can also be caused by a trusted client which just had a bad software update.

AlinMoldovean, 2019-11-05:
@DanielZ87,

Any OPC UA client that can connect to a UA Server can produce a specific load on the server side, e.g. by sending frequent service calls or creating a high number of monitored items.
You can avoid this with the internal configuration mechanism by only allowing a list of _trusted_ clients to connect to the server.
But the logic for protecting the server against a _trusted_ client should be implemented at application level, e. g. by disconnecting those clients.

DanielZ87, 2019-11-05:
Understood. Thx for the help.

closed 2019-11-05 by DanielZ87