OPCFoundation/UA-.NETStandard #850: How can user write nodes with AccessLevel CurrentRead

Nik654, 2019-11-18:
Hello,

I have a question regarding usage of client part of Stack.

On my OPC server i have some nodes which have AccessLevel CurrentRead and UserAccessLevel CurrentRead. 
I have created OPC client using OPC UA.NETStandard Stack and i have used OnServerStarted overridden method (from StandardServer class)
server.SessionManager.ImpersonateUser += new ImpersonateEventHandler(OnSessionLogin);
to check the identity (username/password) to accept/reject users.

**I have a question regarding UserAccessLevel, when i identify that correct user has logged in, how can i give him all the privileges e.g. most important for the user is to write nodes with AccessLevel CurrentRead and all other priviliges like adding nodes etc.
Can this be done and how could i do it?**

I know i can override ValidateRequest and this method i have used not to allow certain operations for the Anonymous user, but i don't see i way have can i use this to allow some users to write nodes which are not writable (have AccessLevel CurrentRead)

fchaxel, 2019-11-21:
Hi,

I've make something like that.

Got some BaseDataVariableState.

All of them have the AccessLevel property set to Read only by default.

On some variables the event handler is set OnReadAccessLevel = OnReadUserAccessLevel;

With something like this

`
public ServiceResult OnReadUserAccessLevel(ISystemContext context, NodeState node, ref byte value)
        {
            value = AccessLevels.CurrentRead;

            if ((context.UserIdentity != null)&&(context.UserIdentity.DisplayName=="myuserwithwriteaccess"))
            {
                value = AccessLevels.CurrentReadOrWrite;
            }

            return ServiceResult.Good;
        }`

So with this the AccessLevel is well displayed for the user with write access.

Then in the OnWrite event Handler I've check again the context.UserIdentity

` if ((context.UserIdentity == null) || (context.UserIdentity.DisplayName != "myuserwithwriteaccess"))
                        return new ServiceResult(StatusCodes.BadUserAccessDenied, new LocalizedText("BadUserAccessDenied"));`

Hope it's helpful.

Bye.

AlinMoldovean, 2019-11-21:
Hi @Nik654 ,

A good example can be found in UserAuthentication Client and Server samples provided in this repository at:
https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Workshop/UserAuthentication

closed 2020-03-17 by AlinMoldovean