OPCFoundation/UA-.NETStandard #875: Future of the prototyping-pubsub branch

romatthe, 2019-12-24:
We've been doing research for a OPC UA PubSub PoC at my company. During the research, we found that this repository contains a stale branch called `prototyping-pubsub`, containing an implementation of the specification. While I've not been able to verify how complete this implementation is exactly, it does contain working (albeit slightly buggy) samples for showing off the PubSub functionality.

Considering that there are currently no open PRs on this branch, I've been wondering if there are any plans of merging this functionality into master, and whether there is any place where we can actually track that effort? It seems to me like a lot of work has gone into the current state of the implementation of that branch, and it would seem like a shame to see it go to waste.

Does anybody have any info on this?

mpostol, 2019-12-25:
@romatthe I have implemented pubsub and now I try to trace any changes in the mpostol/OPC-UA-OOI#208. Hope you will find there something interesting.
I will be back with more details next week, as this topic is on my roadmap too.

mpostol, 2020-01-06:
Hi All,

@romatthe,  let me return to this topic. I have gathered some facts.

1. Description of the PubSub you can find [here](https://commsvr.gitbook.io/ooi/reactive-communication/readme.pubsubmtf)
1. PubSub can and shall be implemented as a loosely coupled component - proof of concept is [here](https://commsvr.gitbook.io/ooi/reactive-communication/semanticdata) and available as the [NuGet package](https://www.nuget.org/packages/UAOOI.Networking.SemanticData)
1. The [HMI on Xamarin](https://github.com/Drutol/CrossHMI) is proof of concept how to use this NuGet
1. Another example you may find [here](https://commsvr.gitbook.io/ooi/reactive-communication/referenceapplication)
1. This implementation is capable to publish data from any source. There are two examples: [(a) Test Data Generator](https://commsvr.gitbook.io/ooi/reactive-communication/referenceapplication/simulatorinteroperabilitytest) and [(b) Boilers Set Simulator](https://commsvr.gitbook.io/ooi/reactive-communication/referenceapplication/simulator.boiler)
1. There is a plan to implement a couple with the generic communication hub called [Process Observer](https://github.com/mpostol/ProcessObserver) supporting **Machine to Sensors Connectivity (OOI.M2S)**
1. Recently the document OPC 10001-6 - Amendment 6: UADP Header Layouts (PDF/ZIP) was published so there is no need to postpone the implementation anymore
1. There are many problems reported for the specification but using it I defined the `NetworkMessage` syntax using Augmented BNF (ABNF). The ABNF syntax definition is [here](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/MessageHandling/NetworkMessage.abnf)
1. There is also work conducted on `Object-Oriented Internet - reactive visualization of asynchronous data using AZURE services` in the project [Networking.Gateway2Azure](https://github.com/mpostol/OPC-UA-OOI/projects)

## Conclusion

Consider reuse of existing implementation - it is free. In case something must be improved I am open to any suggestions and contributions. The work will be conducted on the foundation of the [Partnership Program](https://github.commsvr.com/AboutPartnershipProgram.md.html), which I have just launched. It means that the community will decide about the requirements priority. Consider joining.

ckmk14, 2020-01-12:
Hi, I'm also wondering if it is already possible to estimate when it will be merged?

MaartenBE, 2020-02-04:
@mpostol Thanks for your input and what looks like interesting work in your OOI project. For me it is not very clear what the relationship is between the OOI repository and this repository. Is your work based on the pub/sub prototyping branch? If so, is it your intention to create a pull request to introduce pub/sub to master at some point?

Maybe one of the main contributors of the UA- .NETStandard project can comment on the future of the stale pub/sub branch.

mpostol, 2020-02-04:
@MaartenBE thanks for the feedback. My point is as follows:

- The deliverables, (e.g. NuGet packages) from the OOI repository may be used by this repository.
-  Adding this dependency must be agreed with the maintainers of this repository
-  Implementation of this requires that the API of the OOI code must be compliant with the requirements of this code.

All of that requires joined effort.
Hope it helps,

AlinMoldovean, 2020-11-06:
Moved to #1169

closed 2020-11-06 by AlinMoldovean