OPCFoundation/UA-.NETStandard #876: [question] Browsing Nodes - Sample Console Clients

captmomo, 2019-12-26:
Hi,

I would like to know how to find out all the node ids available on a server as fast as possible.

I've tried the Sample Console clients and have a few questions:

1. Why does the BrowseAllVariables() for the NetCoreComplexClient take so long to find the nodes for some servers (e.g. opc.tcp://opcuaserver.com:48484), but is quick for a sample server running on my local host?
https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/Samples/NetCoreComplexClient/Program.cs

2. What is this code doing?

```
 ReferenceDescriptionCollection references;

            Byte[] continuationPoint;

            references = session.FetchReferences(ObjectIds.ObjectsFolder);

            session.Browse(

                null,

                null,

                ObjectIds.ObjectsFolder,

                0u,

                BrowseDirection.Forward,

                ReferenceTypeIds.HierarchicalReferences,

                true,

                (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,

                out continuationPoint,

                out references);

            Console.WriteLine(" DisplayName, BrowseName, NodeClass");

            foreach (var rd in references)

            {

                Console.WriteLine(" {0}, {1}, {2}", rd.DisplayName, rd.BrowseName, rd.NodeClass);

                ReferenceDescriptionCollection nextRefs;

                byte[] nextCp;

                session.Browse(

                    null,

                    null,

                    ExpandedNodeId.ToNodeId(rd.NodeId, session.NamespaceUris),

                    0u,

                    BrowseDirection.Forward,

                    ReferenceTypeIds.HierarchicalReferences,

                    true,

                    (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,

                    out nextCp,

                    out nextRefs);

                foreach (var nextRd in nextRefs)

                {

                    Console.WriteLine("   + {0}, {1}, {2}", nextRd.DisplayName, nextRd.BrowseName, nextRd.NodeClass);

                }

            }
```

Is it getting all the nodes then iterating through the child nodes?

Sorry for all the questions and thank you for your help!

AlinMoldovean, 2020-01-07:
Hi @captmomo ,

OPC UA Clients cannot retrieve all the nodes of a server in a single service call.
An option for this is to perform a recursive Browse to discover the complete address space of a server, but that can be a time consuming operation in case the server has a high number of nodes.

_"BadAttributeIdInvalid"_ error code is returned by the server for a read call in case the specified attribute is not valid for the specified NodeId.  In case of _Session.ReadValue(NodeId nodeId)_ this should happen in case the specified node is not a Variable node.

captmomo, 2020-01-12:
Thank you. Why does the ComplexClient browse nodes function, BrowseAllVariables(), use a different browsing logic from the SimpleClient? What extra information does it provide?

AlinMoldovean, 2020-01-28:
Hi @captmomo ,

BrowseAllVariables() provides an example of a recursive Browse to get all the Variable nodes of the server.

closed 2020-03-05 by mregen