OPCFoundation/UA-.NETStandard #878: The call is ambiguous between the following methods or properties

SebZar, 2020-01-03:
Hi,
when I include the package OPCFoundation.NetStandard.Opc.Ua in a ASP .NET Core Web Application (.NET Core 3.1), I get the following error when I want to configure Kestrel to use HTTPS.

```
using Microsoft.AspNetCore.Hosting;
using Microsoft.Extensions.Hosting;

namespace MyNamespace {
    public class Program {
        public static void Main(string[] args) {
            CreateHostBuilder(args).Build().Run();
        }

        public static IHostBuilder CreateHostBuilder(string[] args) =>
            Host.CreateDefaultBuilder(args)
                .ConfigureWebHostDefaults(webBuilder => {
                    webBuilder.UseStartup<Startup>()
                    .ConfigureKestrel(serverOptions => {
                        serverOptions.ListenAnyIP(443, listenOptions => {
                            listenOptions.UseHttps("MyCert.pfx", "MyPassword");
                        });

                    });
                });
    }
}
```

Error:

`The call is ambiguous between the following methods or properties: 'Microsoft.AspNetCore.Hosting.ListenOptionsHttpsExtensions.UseHttps(Microsoft.AspNetCore.Server.Kestrel.Core.ListenOptions, string, string)' and 'Microsoft.AspNetCore.Hosting.ListenOptionsHttpsExtensions.UseHttps(Microsoft.AspNetCore.Server.Kestrel.Core.ListenOptions, string, string)'`

Can you please take a look.

Thanks in advance
Sebastian

mregen, 2020-01-08:
Hi @SebZar , could you check if this is related to #598 and fixed by #882?

SebZar, 2020-01-09:
Hi @mregen ,
The fix #882 works, but the mentioned update of the nuget spec must also done to fix the problem.

Pull Requests: 
https://github.com/OPCFoundation/UA-.NETStandard/pull/882
https://github.com/OPCFoundation/UA-.NETStandard/pull/883

Thanks

closed 2020-01-09 by SebZar