OPCFoundation/UA-.NETStandard #886: how to use login/password to do one account can read/write another account only read?

acer1204, 2020-01-15:
I download the UA-.NETStandard and build the opc ua server,

I want to create two account user,One is admin(read/write), another is user(only read) .

if two user login admin and user in same time.

one will show all nodeid level read/write, another will show nodeid level read only.

how to do that like siemens opc ua server?

AlinMoldovean, 2020-01-15:
Hi @acer1204 ,

You can have a look at [UserAuthentication ](https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Workshop/UserAuthentication/Server) sample projects of this repository.

acer1204, 2020-01-21:
> Hi @acer1204 ,
> 
> You can have a look at [UserAuthentication ](https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Workshop/UserAuthentication/Server) sample projects of this repository.

thank you.

i read the .cs file, and try to do this work.

closed 2020-03-17 by AlinMoldovean