OPCFoundation/UA-.NETStandard #888: Different Date Format between LDS and OPC UA server issue

STUBA-rupr, 2020-01-18:
I observed that I get the **BadCertificateTimeInvalid** error when I have at LDS side the US date format M/d/yyyy and at OPC UA server side I have UK date format dd/MM/yyyy.
Both certificate were generated automatically, I was not using any tool to generate certificates.
Most likely it is not a real issue.
PS: Both LDS (https://github.com/OPCFoundation/UA-LDS) and OPC UA Server (https://github.com/OPCFoundation/UA-.NETStandard) are from OPC Foundation git.
BR Rudo

AlinMoldovean, 2020-01-29:
Hello @STUBA-rupr ,

The **BadCertificateTimeInvalid** error is generated at the OPC UA .NET Standard Server side during certificate validation in case the certificate is expired or not yet valid. Time format should normally not be affect this.
Could you check if the error is generated at server side or at LDS side?
You also need to check both UA server and LDS certificate for expiration. In this case you need a renewal of the expired certificate.

rudoq32, 2020-02-03:
Hello @AlinMoldovean ,
The error was generated at server side. I deleted all certificates to force renew. It worked for me before.
I guess the root cause is a combination of dd/MM which does not exist in M/d (for an instance 16 days ago was 18th of January, so 18/01 is invalid in M/d).

STUBA-rupr, 2020-02-03:
Hello @AlinMoldovean ,
The error was generated at server side. I deleted all certificates to force renew. It worked for me before.
I guess the root cause is a combination of dd/MM which does not exist in M/d (for an instance 16 days ago was 18th of January, so 18/01 is invalid in M/d).

closed 2020-03-05 by mregen