OPCFoundation/UA-.NETStandard #890: Vantage Technology Corp updates to 1.01 core  DotNet Stack source code

mklein9, 2020-01-23:
This issue is opened in order to upload core files Vantage Technology Corp. has modified in adherence to the RCBL 1.00 license they were originally downloaded under. The base source code is revision 1.01, from April 2012, and so is not contained in this git repo. On advice from Directory of Compliance Paul Hunkar 01/23/2020, this issue is opened and modified files attached in order to fulfill Vantage Technology Corp's obligations under RCBL 1.00. The RCBL may be viewed at [http://opcfoundation.org/License/RCBL/1.00/](http://opcfoundation.org/License/RCBL/1.00/).

The changes include:

* Server window starts minimized and with the Close button disabled to reduce chance of accidental exit
* Build script change to ensure target directory exists before copying files into it
* Component and project file updates to support .NET 4.5, Visual Studio 2013, and Windows 10

The solution file is also included, but please note it references a few project files that are Vantage specific but are under the MIT license, so are not included here. Use of this solution file will require some minor changes to restore the reference to the sample projects.
[upload.tar.gz](https://github.com/OPCFoundation/UA-.NETStandard/files/4106173/upload.tar.gz)

dk2129, 2020-01-24:
Dear Mike,
Thank you for providing this. This issue applies to an older version of the SDK. For this version we have a so-called RCL Sharepoint where all contributions are kept. I pushed your contribution to this Sharepoint in a "VantageTechnologyCorp" folder.
Your obligations are fulfilled.

The moderators of this repository should check your contribution to see if some elements still apply.

Best Regards,
Karl Deiretsbacher
Technical Director, OPC Foundation

mklein9, 2020-01-24:
Karl,

Thank you very much for doing the push and letting me know the status.

Best regards,

		-Mike

closed 2020-01-24 by mklein9