OPCFoundation/UA-.NETStandard #891: Call OPC Ua Method with complex input type

LukasSchwendemann, 2020-01-24:
Hey there,

I have an issue using the client API to make a call of an OPC UA function with a complex type as input. It is no Problem to give a struct as input.

The complex type I want to use is generated in runtime from the type information of the OPC UA Server. Therefore, I can not instantiate the needed type and assign the necessary values in code.

I tried to use anonymous types. But using them I get an Exception that the type is not convertible into a Variant.

Example:

```
var session = new Session(testChannel, testConfiguration, testEndpoint);
var inputObj = new { TestProperty = 1, TestProperty2 = "TestInput" };
var inputString1 = "inputString1";
var inputString2 = "inputString2";
var callResult = session.Call(testNodeId, testMethodId, inputString1, inputObj, inputString2);
```

Can someone help me ho to structure the input to make a successful Call with a complex input type?

mregen, 2020-01-24:
@LukasSchwendemann Hi Lukas, did you look at the NetCoreComplexClient? You can load the types, but then to instantiate the needed type use the same mechanism as the encoders, create it by NodeId. The complex types created at runtime have accessors which you can use to set parameters. see sample and let me know if you need more info...

amzex, 2020-03-17:
Hi @mregen !
Can you give us an example of how to use NetCoreComplexClient? 

Thanks!

mregen, 2020-03-18:
Crazy days, will do when I have some cycles ...

amzex, 2020-05-20:
Hi @mregen!
Can you help me use the NetCoreComplex Client library?
I couldn't move forward with my project.
Thanks!

AlansZZm, 2020-06-13:
Hi @mregen 
I also meet the same question like lukas .
The input type is :

<opc:TypeDictionary
  xmlns:opc="http://opcfoundation.org/BinarySchema/"
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns:ua="http://opcfoundation.org/UA/"
  xmlns:tns="http://opcfoundation.org/UA/DI/"
  DefaultByteOrder="LittleEndian"
  TargetNamespace="http://opcfoundation.org/UA/DI/"
>
  <opc:Import Namespace="http://opcfoundation.org/UA/" Location="Opc.Ua.BinarySchema.bsd"/>

  <opc:StructuredType BaseType="ua:ExtensionObject" Name="KeyIntPair">
    <opc:Field TypeName="opc:String" Name="Key"/>
    <opc:Field TypeName="opc:Int32" Name="Value"/>
  </opc:StructuredType>

 
</opc:TypeDictionary>

I want to use " public object[] CallMethodByNodeId(string tagParent, string tag, params object[] args);"to call the method . I do not know how to make the  params object[] args to let the robot get it . For example ,if I want to send (test_out as the key ,666 as the value ) to the robot ,how to make 
my "args".
[image]

Jetsord, 2020-12-07:
Hello there,

I made a simple solution with the Opc.Ua.Client to call a method with a input struct (without using the ComplexClient).

I added this following part to the Session.cs (of the Opc.Ua.Client)

This is my (full) Session.cs (https://github.com/OPCFoundation/UA-.NETStandard/files/5653923/Session.cs.txt)
(Session.cs from Release 1.4.363.107 + my new "CallStruct"-method)

        /// <summary>
        /// Calls the specified method and returns the output arguments.
        /// </summary>
        /// <param name="objectId">The NodeId of the object that provides the method.</param>
        /// <param name="methodId">The NodeId of the method to call.</param>
        /// <param name="dataValue">The input argument in a structure </param>
        /// <returns>The list of output argument values.</returns>
        public IList<object> CallStruct(NodeId objectId, NodeId methodId, ExtensionObject dataValue)
        {
            VariantCollection inputArguments = new VariantCollection();

            inputArguments.Add(new Variant(dataValue));

            CallMethodRequest request = new CallMethodRequest();

            request.ObjectId = objectId;
            request.MethodId = methodId;
            request.InputArguments = inputArguments;

            CallMethodRequestCollection requests = new CallMethodRequestCollection();
            requests.Add(request);

            CallMethodResultCollection results;
            DiagnosticInfoCollection diagnosticInfos;

            ResponseHeader responseHeader = Call(
                null,
                requests,
                out results,
                out diagnosticInfos);

            ClientBase.ValidateResponse(results, requests);
            ClientBase.ValidateDiagnosticInfos(diagnosticInfos, requests);

            if (StatusCode.IsBad(results[0].StatusCode))
            {
                throw ServiceResultException.Create(results[0].StatusCode, 0, diagnosticInfos, responseHeader.StringTable);
            }

            List<object> outputArguments = new List<object>();

            foreach (Variant arg in results[0].OutputArguments)
            {
                outputArguments.Add(arg.Value);
            }

            return outputArguments;
        }
         

So now I can call a method with 1 or more structs as an input!

The following parts you have to add to your program (after the connection handling):
In my simple solution you can not mix-up the input types like "string, int, struct", only 1 struct or 1 array of structs,
because I need everytime only 1 struct with inner (mixed-up) datatypes.

For calling a simple method with plain datatypes like string, int, ..., you can use the Method "CallMethod" instead of "CallStructMethod"

This is my example for calling this function and the InputData looks like this:
StructName    |    ..        |  STRUCT/UDT:StructName
Item1              |   Hello  | String
Item2              |   1337  |  Int32
ItemX
StructName    |    ..        |  END_STRUCT/UDT

[UAClientAPI.txt](https://github.com/OPCFoundation/UA-.NETStandard/files/5654057/UAClientAPI.txt)

         //Exampe Code for calling a method with a struct
         public IList<object> CallStructMethod(String objectIdString, List<string[]> inputData)
         {
             //For calling a method we need it's node id and it's parent object's node id
              NodeId methodNodeId = new NodeId(objectIdString + ".Method");
             NodeId objectNodeId = new NodeId(objectIdString);

            //Create an ExtensionObject from the Structure given to this function   
            ExtensionObject writeExtObj = new ExtensionObject();

            //This is for regular Struct/UDT            
            if (!inputData[0][2].Contains("OF STRUCT/UDT"))
            {
                //Determine lentgh of byte array needed to contain all data
                Int64 length = GetLengthOfDataToWrite(inputData);

                //Create a byte array
                byte[] bytesToWrite;

                //Parse dataToWrite to the byte array
                bytesToWrite = ParseDataToByteArray(inputData, length);

                //Copy data to extension object body
                writeExtObj.Body = bytesToWrite;

            }
            //This is for array of Struct/UDT
            else
            {
            ...
            }
            //Declare a list of objects for the method's output arguments
            IList<object> outputArguments = new List<object>();

            //Call the method
            outputArguments = mSession.CallStruct(objectNodeId, methodNodeId, writeExtObj);

            return outputArguments;
        }

@mregen 
Hello Martin,
it would be great if you can merge my "CallStruct" (in Session.cs) to the offical (Opc.Ua.Client) project, I think it is not only for me a usefull addon.

Kind Regards Julian

AlinMoldovean, 2021-03-16:
Closed in #1236

closed 2021-03-16 by AlinMoldovean