OPCFoundation/UA-.NETStandard #892: JsonDecoder parsing is dependent on current culture

ZbynekZ, 2020-01-24:
At multiple places in JsonDecoder.cs, numbers are parsed with .TryParse method overload that uses the current thread's culture for parsing, and not a fixed invariant culture.

mregen, 2020-01-25:
Hi @ZbynekZ, could you check if this is the right approach [here](https://github.com/mregen/UA-.NetStandardLibrary/tree/encodingtests). I'm working on #874 and I'm also adding tests. Would be great if you could take a look at the fixes too. Thanks, Martin

ZbynekZ, 2020-01-26:
Yes, I think it is corrected now. The only thing I do not understand is that why in ReadUInt64, there is NumberStyles.AllowLeadingWhite | NumberStyles.AllowTrailingWhite, while parsing other integer types uses NumberStyles.Integer, Unless there is something special about UInt64 that I do not know, and assuming that the tokens come in already stripped off the whitespace, I think that NumberStyles.Integer should be used with UInt64 as well.

mregen, 2020-01-26:
Hi @ZbynekZ, my idea was just to disallow the leading sign. But sign can also be positive so I better go with Integer as you suggested. Thanks, Martin

closed 2020-02-12 by mregen