OPCFoundation/UA-.NETStandard #896: BadTCPInternalError - iot opc publisher

marru8619, 2020-01-31:
Hi, 

I'm trying to connecto to an OPC Ua Server with Basic256 and Sign&Encrypt Security Policy. The certificates are generate correctly and copied in the trust folder (server and client). The problem is that when I start the program that error coming up:

2020-01-31 12:33:46.506 +01:00 [INF] Create secured session for endpoint URI 'opc.tcp://192.168.13.101:49320' with timeout of 50000 ms.
2020-01-31 12:33:47.170 +01:00 [ERR] Session creation to endpoint 'opc.tcp://192.168.13.101:49320' failed 7 time(s). Please verify if server is up and Publisher configuration is correct.
System.AggregateException: One or more errors occurred. (Error establishing a connection: Could not send an Open Secure Channel request.) ---> Opc.Ua.ServiceResultException: Error establishing a connection: Could not send an Open Secure Channel request.
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 190
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 338
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 344
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 294
   at Opc.Ua.SessionClient.CreateSession(RequestHeader requestHeader, ApplicationDescription clientDescription, String serverUri, String endpointUrl, String sessionName, Byte[] clientNonce, Byte[] clientCertificate, Double requestedSessionTimeout, UInt32 maxResponseMessageSize, NodeId& sessionId, NodeId& authenticationToken, Double& revisedSessionTimeout, Byte[]& serverNonce, Byte[]& serverCertificate, EndpointDescriptionCollection& serverEndpoints, SignedSoftwareCertificateCollection& serverSoftwareCertificates, SignatureData& serverSignature, UInt32& maxRequestMessageSize) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 108
   at Opc.Ua.Client.Session.Open(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain) in C:\Build\Projects\NETStandard-NuGet\src\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 2372
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales) in C:\Build\Projects\NETStandard-NuGet\src\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 953
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at OpcPublisher.OpcUaSession..ctor(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales) in C:\ProgettiMariano\iot-edge-opc-publisher-master\iot-edge-opc-publisher-master\opcpublisher\OpcUaSession.cs:line 16
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in C:\ProgettiMariano\iot-edge-opc-publisher-master\iot-edge-opc-publisher-master\opcpublisher\OpcSession.cs:line 459
---> (Inner Exception #0) Opc.Ua.ServiceResultException: Error establishing a connection: Could not send an Open Secure Channel request.
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 190
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 338
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 344
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 294
   at Opc.Ua.SessionClient.CreateSession(RequestHeader requestHeader, ApplicationDescription clientDescription, String serverUri, String endpointUrl, String sessionName, Byte[] clientNonce, Byte[] clientCertificate, Double requestedSessionTimeout, UInt32 maxResponseMessageSize, NodeId& sessionId, NodeId& authenticationToken, Double& revisedSessionTimeout, Byte[]& serverNonce, Byte[]& serverCertificate, EndpointDescriptionCollection& serverEndpoints, SignedSoftwareCertificateCollection& serverSoftwareCertificates, SignatureData& serverSignature, UInt32& maxRequestMessageSize) in C:\Build\Projects\NETStandard-NuGet\src\Stack\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 108
   at Opc.Ua.Client.Session.Open(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain) in C:\Build\Projects\NETStandard-NuGet\src\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 2372
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales) in C:\Build\Projects\NETStandard-NuGet\src\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 953<---

Someone know how to fix?

mregen, 2020-01-31:
Hi @marru8619, I see this `C:\ProgettiMariano\iot-edge-opc-publisher-master\iot-edge-opc-publisher-master\opcpublisher\OpcUaSession.cs:line 16` line in your log, which application is this? Do you have a link? Also which nuget version are you using?

marru8619, 2020-01-31:
Hi mregen,

the program is here:
https://github.com/Azure/iot-edge-opc-publisher/tree/master

The nuget version of OPC UA Foundation is the last. The OPC UA server is a Siemens. 
The problem is that, while the program generate the certificate, the server doesn't receive the request to trust the certificate.

mregen, 2020-01-31:
@marru8619 did this work in the previous nuget package?

marru8619, 2020-01-31:
I never tested the previous nuget package with an opc server with basic256 and sign&Encrypted securtity policy.

mregen, 2020-02-01:
Hi @marru8619 , I edited your comment to make it readable. Looks like you endpoint Uri is not quite right or not reachable.

marru8619, 2020-02-04:
Hi mregem,

I have found the problem. The private certificate needed to be imported on the host machine. 

There is another question: If the OPC Server is configured with an username and password the publisher goes fail and the program is close. 

That because the program acept only an encrypted user name and password passed through the publishednode configuration file:

"OpcAuthenticationMode" : "UsernamePassword",
"OpcAuthenticationUser" : "<Username>",
"OpcAuthenticationPassword" : "<Password>",

as written here:
https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-publisher-configure

but there is a problem in the documentation:

the Username and Password need to be passed written in basic64-string mode:

"EncryptedAuthUsername":"<UsernameBasic64-String>",
"EncryptedAuthPassword":"<PasswordBasic64-String>",

the think is why?

mregen, 2020-02-06:
Hi @marru8619 , please open and discuss publisher issues in the repo of publisher. Here is just for the .Net stack. The latest version of the publisher you can find here: https://github.com/Azure/Industrial-IoT/tree/release/2.5.3
The latest version is still in preview. Thanks, Martin

closed 2020-02-06 by mregen