OPCFoundation/UA-.NETStandard #900: OPC-UA-Client Session.Create raises exception

jvanlangen, 2020-02-06:
I'm having trouble implementing an authentication on an OPC UA client. It works without anthentication, but providing a Username/Password a `System.NullReferenceException: 'Object reference not set to an instance of an object.'` raises.

I think it's a shame when a third party component raises a NullReferenceException, so I presume it is a bug.

My knowledge about user authentication/certificates is nearly zero.

I've added the `new UserTokenPolicy(UserTokenType.UserName)` but I'm ignorant how it works.

Here's the code I use:

    Utils.SetTraceOutput(Utils.TraceOutput.Off);
    var config = new ApplicationConfiguration()
    {
        ServerConfiguration = new ServerConfiguration
        {
            UserTokenPolicies = new UserTokenPolicyCollection(new[] { new UserTokenPolicy(UserTokenType.UserName) }), 
        },
        ApplicationName = "MyHomework",
        ApplicationType = ApplicationType.Client,
        SecurityConfiguration = new SecurityConfiguration
        {
            ApplicationCertificate = new CertificateIdentifier
            {
                StoreType = @"Windows",
                StorePath = @"CurrentUser\My",
                SubjectName = Utils.Format(@"CN={0}, DC={1}", "MyHomework", System.Net.Dns.GetHostName())
            },
            TrustedPeerCertificates = new CertificateTrustList
            {
                StoreType = @"Windows",
                StorePath = @"CurrentUser\TrustedPeople",
            },
            NonceLength = 32,
            AutoAcceptUntrustedCertificates = true
        },
        TransportConfigurations = new TransportConfigurationCollection(),
        TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
        ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000, }
    };
    config.Validate(ApplicationType.Client);
    if (config.SecurityConfiguration.AutoAcceptUntrustedCertificates)
    {
        config.CertificateValidator.CertificateValidation += (s, e) => { e.Accept = true; };
    }

    Session = Session.Create(config, new ConfiguredEndpoint(null, new EndpointDescription(url)), true, "", 60000, new UserIdentity("user", "pwd"), null);

AlinMoldovean, 2020-02-10:
Hi @jvanlangen ,

Please have a look at [UserAuthentication](https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Workshop/UserAuthentication) sample projects as an example of uer authentication implementation.

mregen, 2020-03-05:
The StoreType Windows doesn't exist in .NetStandard. Is this the .Net legacy stack?

closed 2020-03-17 by AlinMoldovean