OPCFoundation/UA-.NETStandard #901: About .net UA server official Certification question.

acer1204, 2020-02-10:
In github project(.net standard). it say pass "this project has been certified for compliance through an OPC Foundation Certification Test Lab to prove its high quality". 

When i use this github project to do my work. I add robot control communication, that i can use opc ua client to control my robot. 
My company have join the Class D member, If my company want to sell this opc ua server, I need to official Certification again?

eoursel, 2020-02-10:
@acer1204 no this is not mandatory. You don't need to certify your product again albeit this is highly recommended to do it
Regards

ZbynekZ, 2020-02-10:
I think the answer from @eoursel is incomplete and can be greatly misunderstood.

First of all, the question asks " If my company want to sell this opc ua server, I need to official Certification again?". The answer to it should be divided into two cases:

a) You do not need *any* official certification, if you do not want to claim that you have it. So, you can sell any OPC UA server without certification - you just cannot say that it *is* certified in that case.

b) If you want to claim that the server *is* certified, then of course it has to be truly certified. In this case, what is certified is precisely the code that resides in GitHub, and not anything that you derive from it. Therefore, if you, as you wrote "add robot control communication" (to the server code), and want to claim that server as OPC UA certified, your company will have to go through the OPC certification process with it.

acer1204, 2020-02-10:
@ZbynekZ

thanks a lot. 

I download the official Certification UACTT tool. and UA-.NET-Legacy project.

When I running  the ua server and try to use the UACTT tool to browser the server node.

It always show Error loading ClientCertificate: BadInvalidArgument(0x80ab0000)...

I try to copy the UACTT ClientCertificate opcuactt.der to ua server MachineDefault path, but it always shot the error.

How can i solve this problem?

[image]
[image]

AlinMoldovean, 2020-02-10:
Hi @acer1204 ,

The issue seems to be on the Compliance Test Tool side.
It seems that the application instance certificate cannot be loaded for some reason.

acer1204, 2020-02-10:
@AlinMoldovean 

I download the UA-.NETStandard, but it compile failed.

it show some error about .net 4.6 dll.
[image]

acer1204, 2020-02-11:
@AlinMoldovean

I try to use in two project.

The UACTT can't test UA-.NET-Legac project, but it can test UA-.NETStandard project.

When I use UA-.NETStandard and test it like UACTT server test video, it working!!

Thank.

closed 2020-02-11 by acer1204