OPCFoundation/UA-.NETStandard #903: Binding Xamarin Client

torikachi24, 2020-02-15:
I want to add binding in Xamarin Client but it not working
-**MainPage.xaml**
>     <ContentPage xmlns="http://xamarin.com/schemas/2014/forms"
>                 xmlns:x="http://schemas.microsoft.com/winfx/2009/xaml"
>                 xmlns:local="clr-namespace:XamarinClient"
>                 x:Class="XamarinClient.MainPage"
>                 Title="OPC UA Client">
>        <ContentPage.BindingContext>
>            <local:ConnectionListViewModel/>
>        </ContentPage.BindingContext>
>        <ContentPage.Content>
>            <StackLayout  Margin = "20,20,20,20">
>                <StackLayout VerticalOptions="FillAndExpand" HorizontalOptions="Fill">
>                    <Label Text="OPC UA server's endpoint URL:" FontSize="Medium" TextColor="#0066ff" FontAttributes="Bold"/>                       
>                    <Entry x:Name="EntryUrl" Placeholder="Write your EndpointUrl here" Text="{Binding ConnectionName}"/>
>                 <ListView x:Name="ContactsList" ItemsSource="{Binding Connections}" >     `
> 
 -**ConnectionListViewModel:**

>    namespace XamarinClient
>   {
>         public class ConnectionListViewModel:INotifyPropertyChanged
>    {
>        public ICommand AddConnectionNameCommand => new Command (AddConnectionName);
>       public ObservableCollection<string> Connections { get; set; }
>       public string ConnectionName { get; set; }
>       public string ConnectionUrl { get; set; }
>       public ConnectionListViewModel()
>       {
>               Connections = new ObservableCollection<string>();
>               Connections.Add("text");
>       }
>       public void  AddConnectionName()
>       {
>               Connections.Add(ConnectionName);
>       }
>       public event PropertyChangedEventHandler PropertyChanged;
>       private void OnPropertyChanged(string propertyName)
>       {
>               if (this.PropertyChanged != null)
>                   this.PropertyChanged(this, new PropertyChangedEventArgs(propertyName));
>       }
>       }
>   }   

- And the result. I have tried in new project ,Binding is good...but in UA Xamarin Client it not working and show problem in debug.Do you know what is the problem? Thanks
` [0:] Binding: 'ConnectionName' property not found on 'XamarinClient.ConnectionListViewModel', target property: 'Xamarin.Forms.Entry.Text'  `

idSoftwareWizard, 2020-03-25:
To help diagnose this you need to paste your MainPage.xaml.cs too.

christhiel, 2020-03-30:
@torikachi24 
You bind the ItemSource to Connections, but this is an untyped ObservableCollection, that couldn´t work.
First, you should create a Model with Properties like this:
`public class Connection`
`{`
   `    public string ConnectionName { get; set; }`
   `    public string ConnectionUrl { get; set; }`
`}`

In your ViewModel create a generic ObservableCollection<T> Property like this:
`private ObservableCollection<Connection> connections;`
`    public ObservableCollection<Connection> Connections`
    `{`
        `get { return connections;}`
        `set `
        `{ `
            `connections = value;`
            `OnPropertyChanged(Connections);`
`        }`
`    }`
You need a full Property with a backing field and you need to call OnPropertyChanged().
Then you could add a new Connection to your ObservableCollection<Connection> Connections.

Hope, it helps.