OPCFoundation/UA-.NETStandard #914: NetCoreComplexTypeClient and MachineVision

eoursel, 2020-03-07:
hi @mregen 

I am trying to test your complex type client with a OPC UA server with a Machine Vision information model which contain many complex types. Only a few of them are resolved. I am trying to understand what's is wrong.  Did you test this kind of configuration with a complex information model such as the VDMA Machine Vision information model?

mregen, 2020-03-07:
hi @eoursel, I'm attending the IOP next week so hopefully we get better coverage. There could be multiple reasons for failure, is it a DataTypeDefinition based server (V1.04) or dictionary based (V1.03)? Could you share a logfile and more info on the failing types? 
Thanks

eoursel, 2020-03-07:
Thanks @mregen, good point. The server is a node opcua server in NodeJS. So it is possible we have an issue on the server, i am checking that. Yes i am working to produce a detailed logfile. 

I was not aware there was an IOP next week. If it is useful for the community, i could make this server accessible on the internet if you want to do some testing at the IOP. 

Thanks

eoursel, 2020-03-08:
Hi @mregen 

Just starting to understand what's wrong. 

The machine vision specification defines a DataType TrimmedString. 

```
    <UADataType NodeId="ns=1;i=3017" BrowseName="1:TrimmedString">
        <DisplayName>TrimmedString</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=12</Reference>
        </References>
    </UADataType>

```
which is a subtype of a string and this subtype is used as the data type of some structured data types. As a consequence all these structured types are not loaded correctly. If you replace this TrimmedString but a String (i=12) it's much better. I have still a few types which not correctly loaded.

That's an interesting subject for the IOP. Sorry it was not possible for me to come. All domestic and international flights are now strictly forbidden by my company because of the COVID-19. 

Regards

mregen, 2020-03-10:
Working on a fix, the server needs to add proper entries for basedatatype and the complex client needs to handle subtypes properly and not fall over with invalid basedatatypes..

mregen, 2020-03-18:
Please try #916

mregen, 2020-03-18:
Fixed with your testserver. Please retry and close issue. 
Thanks

eoursel, 2020-03-18:
@mregen thanks a lot. I will try

closed 2020-04-01 by mregen