OPCFoundation/UA-.NETStandard #919: Asynchronous version of Subscription.Create()

mibos-kremer, 2020-03-13:
Hello,

When registering subscriptions, I am noticing some hiccups when calling `Subscription.Create()` in the Client module (Usually around 200ms per call).

After some profiling, it seems that the `m_session.CreateMonitoredItems()` call (inside CreateItems()) is mainly responsible for this delay. There is an asynchronous version of CreateMonitoredItems, but it does not seem to be possible to use this along with the existing Subscription class.

Would it be possible to get an asynchronous version of `Subscription.Create()`? Or is there another way to speed up the creation of subscriptions?

Sincerly,
Michiel

mregen, 2020-03-20:
Please stay tuned and thanks for #919, we need some time to discuss async API extensions. Thanks

mregen, 2020-06-06:
If we want to be async we should add an API layer for APM/TPL for all public client API.
I started an API prototype based on your #919 here https://github.com/mregen/UA-.NetStandardLibrary/blob/asyncsubscription/SampleApplications/SDK/Opc.Ua.Client/SubscriptionAsync.cs to show how it could look like. 
Once agreed we need to do it for all public API across the board. Need to get feedback if this is a feasible extension in the long run.

mregen, 2020-06-19:
More tbd. Needs Modelcompiler updates https://github.com/mregen/UA-ModelCompiler/tree/no_servicemodel and here is a prototype: https://github.com/mregen/UA-.NetStandardLibrary/tree/servicemod_plc
This is quite a bigger workitem which we need to schedule once repo split is done.

mregen, 2021-01-28:
waiting for #1255

PierluigiCarusoOcima, 2021-05-26:
@mregen any updates?

mregen, 2021-09-06:
Hi @mibos-kremer , sorry for the lengthy delay, finally I prepared a async subscription extension but I'm not planning on APM stubs because there are now the TPL async stubs in core. Do you think this will sufficient to address the problem you were seeing?
--> see #1498

PR #1498 (2021-09-07): Add async versions of subscription API to session client
fixes #919 and #921

- enable async client stubs in core
- add async versions for subscription management
- APM versions will not be included...

based on PR #919 from @mibos-kremer
commit 76afe829 (2021-09-07): Add async versions of subscription API to session client (#1498)

fixes #919 and #921

- enable async client stubs in core
- add async versions for subscription management
- APM versions will not be included...

based on PR #919 from @mibos-kremer
closed 2021-09-07 by mregen