OPCFoundation/UA-.NETStandard #922: Browse the OPC UA server namespace ON SERVER SIDE

barbaragabriella, 2020-03-17:
Hi,

I was wondering if it'd be possible to browse the namespace on the server-side. Or how I could just print it. I'm using the SampleAplication/Sample/NetCoreConsoleServer and NetCoreConsoleClient.

On the client-side, it prints:

```
DisplayName, BrowseName, NodeClass                                                                                      
Server, Server, Object                                                                                      
+ ServerArray, ServerArray, Variable                                                                                    
+ NamespaceArray, NamespaceArray, Variable                                                                              
+ ServerStatus, ServerStatus, Variable                                                                                  
+ ServiceLevel, ServiceLevel, Variable                                                                                  
+ Auditing, Auditing, Variable                                                                                          
+ EstimatedReturnTime, EstimatedReturnTime, Variable                                                                    
+ LocalTime, LocalTime, Variable                                                                                        
+ ServerCapabilities, ServerCapabilities, Object                                                                        
+ ServerDiagnostics, ServerDiagnostics, Object                                                                          
+ VendorServerInfo, VendorServerInfo, Object                                                                            
+ ServerRedundancy, ServerRedundancy, Object                                                                            
+ Namespaces, Namespaces, Object                                                                                        
+ GetMonitoredItems, GetMonitoredItems, Method                                                                          
+ ResendData, ResendData, Method                                                                                        
+ SetSubscriptionDurable, SetSubscriptionDurable, Method                                                                + RequestServerStateChange, RequestServerStateChange, Method                                                            + ServerConfiguration, ServerConfiguration, Object                                                                      
+ PublishSubscribe, PublishSubscribe, Object                                                                            
+ Dictionaries, Dictionaries, Object                                                                                    
+ Boilers, 4:Boilers, Object                                                                                          
Boilers, 4:Boilers, Object,                                                                              
+ Boiler #1, 4:Boiler #1, Object                                                                                        
+ Boiler #2, 5:Boiler #2, Object                                                                                        
+ Boiler #1, 4:Boiler #1, Object                                                                                      
MemoryBuffers, 7:MemoryBuffers, Object, ns=7;i=1025                                                                       + UInt32, 8:UInt32, Object                                                                                              
+ Double, 8:Double, Object
```

But I would be interested in printing the same thing on the server-side. I tried using the Browse method but ``` server.CurrentInstance.DefaultSystemContext.OperationContext``` is null. Please any insight would be very welcome!!

AlinMoldovean, 2020-03-17:
Hi @barbaragabriella ,

Here is a code sample for browsing directly on a Server instance:

```
// Browse the server
BrowseDescriptionCollection nodesToBrowse = new BrowseDescriptionCollection();
BrowseDescription nodeToBrowse = new BrowseDescription();
nodeToBrowse.NodeId = ObjectIds.Server;
nodeToBrowse.ResultMask = (uint)BrowseResultMask.All;
nodesToBrowse.Add(nodeToBrowse);

OperationContext context = new OperationContext(new RequestHeader(), RequestType.Browse);
BrowseResultCollection results = null;
DiagnosticInfoCollection diagnosticInfos = null;

server.CurrentInstance.NodeManager.Browse(
    context,
    null,
    0,
    nodesToBrowse,
    out results,
    out diagnosticInfos);

// Display the results
ReferenceDescriptionCollection nodeReferences = results[0].References;
Console.WriteLine(" DisplayName, BrowseName, NodeClass");
foreach (var reference in nodeReferences)
{
    Console.WriteLine(" {0}, {1}, {2}", reference.DisplayName, reference.BrowseName, reference.NodeClass);
}
```

barbaragabriella, 2020-03-17:
Thank you!!! It helped a lot!!!

May I ask another question? Is there a way I can register a listener to the Subscription.InnerPublish method?  What I want to do is intercept all the outgoing messages so I can do some action before sending them, but I wanted to do it with a method, like when we create a new server and the listeners for the session are passed.
```
// print notification on session events
server.CurrentInstance.SessionManager.SessionActivated += EventStatus;
server.CurrentInstance.SessionManager.SessionClosing += EventStatus;
server.CurrentInstance.SessionManager.SessionCreated += EventStatus;
```

So I wanted to call something like (I guess it'll be from NodeManager) ```server.CurrentInstance.NodeManager.Interceptor += MyEvent;```

AlinMoldovean, 2020-03-20:
Hello @barbaragabriella ,

You can implement overrides in multiple places when using the Server library.
e.g. you can override the Publish service call of _StandardServer_:

```
        public override ResponseHeader Publish(
            RequestHeader requestHeader,
            SubscriptionAcknowledgementCollection subscriptionAcknowledgements,
            out uint subscriptionId,
            out UInt32Collection availableSequenceNumbers,
            out bool moreNotifications,
            out NotificationMessage notificationMessage,
            out StatusCodeCollection results,
            out DiagnosticInfoCollection diagnosticInfos)
        {
            OperationContext context = ValidateRequest(requestHeader, RequestType.Publish);

            try
            {
                Utils.Trace("PUBLISH #{0} RECEIVED. TIME={1:hh:mm:ss.fff}", requestHeader.RequestHandle, requestHeader.Timestamp);

                notificationMessage = ServerInternal.SubscriptionManager.Publish(
                    context,
                    subscriptionAcknowledgements,
                    null,
                    out subscriptionId,
                    out availableSequenceNumbers,
                    out moreNotifications,
                    out results,
                    out diagnosticInfos);

                /*
                 Do something before sending the Publish response
                */

                return CreateResponse(requestHeader, context.StringTable);
            }
            catch (ServiceResultException e)
            {
                lock (ServerInternal.DiagnosticsWriteLock)
                {
                    ServerInternal.ServerDiagnostics.RejectedRequestsCount++;

                    if (IsSecurityError(e.StatusCode))
                    {
                        ServerInternal.ServerDiagnostics.SecurityRejectedRequestsCount++;
                    }
                }

                throw TranslateException(context, e);
            }
            finally
            {
                OnRequestComplete(context);
            }
        }
```

closed 2020-03-30 by AlinMoldovean