OPCFoundation/UA-.NETStandard #930: ApplicationConfiguration.SaveToFile() removes "RejectSHA1SignedCertificates" #184

PrashantDesai, 2020-03-26:
https://github.com/OPCFoundation/UA-.NET-Legacy/issues/184

Is this problem is resolved in following Nuget package?? It is not working for me
1.4.360.33

AlinMoldovean, 2020-03-30:
Hi @PrashantDesai ,

_ApplicationConfiguration.SaveToFile(string filePath)_ writes in the file only the members that have a value different from the _default_ of the type.
This is because _EmitDefaultValue = false_ for all the members of configuration classes.

```
        //
        // Summary:
        //     Gets or sets a value that specifies whether to serialize the default value for
        //     a field or property being serialized.
        //
        // Returns:
        //     true if the default value for a member should be generated in the serialization
        //     stream; otherwise, false. The default is true.
        public bool EmitDefaultValue { get; set; }
```

_RejectSHA1SignedCertificates_ will be written in the file only if its value is "true".

Please note that _SaveToFile()_ produces a slightly different output compared to the initial xml configuration file. e.g it also removes the comments.

closed 2020-03-30 by AlinMoldovean

PrashantDesai, 2020-03-30:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

OK
Thank you

I will check this and let you know

PrashantDesai, 2020-03-30:
We are facing exactly present in this link:
https://opcfoundation.org/forum/opc-ua-implementation-stacks-tools-and-samples/applicationconfiguration-savetofile-removes-rejectsha1signedcertificates/#p1824

If we remove RejectSHA1SignedCertificates in XML the following code gives error as present the above link, then if we put back flag again as false it works fine,

                ApplicationConfiguration configuration = ApplicationConfiguration.Load(new System.IO.FileInfo(configurationFile),
                    Opc.Ua.ApplicationType.Server, null).Result;

                CertificateValidator validator = new CertificateValidator();
                validator.Update(configuration);
                validator.Validate(certificate);

PrashantDesai, 2020-03-30:
we are using 1.4.360.33 nu get package

AlinMoldovean, 2020-03-31:
Hi @PrashantDesai ,

The problem is that you application instance certificate has a SHA1 signature that is not trusted anymore.
If you still want to use it you should do the following:

```
ApplicationConfiguration configuration = ApplicationConfiguration.Load(new System.IO.FileInfo(configurationFile),
                Opc.Ua.ApplicationType.Server, null).Result;

configuration.SecurityConfiguration.RejectSHA1SignedCertificates = false;

            CertificateValidator validator = new CertificateValidator();
            validator.Update(configuration);
            validator.Validate(certificate);

```

PrashantDesai, 2020-03-31:
Hello Alin,

Thanks a lot. It is working.