OPCFoundation/UA-.NETStandard #931: OPC UA Alarm & Conditions not working??

KLMInnovation, 2020-03-27:
I am puzzling for day's now on getting the Alarm & Conditions working.
I have a couple of Sinumerik 840D sl OPC UA enabled machine, where if I use a OPC UA COTS client I immediately receive server Alarm & Conditions.
The Monitoring parameters works fine, receiving DA also works fine, only the Alarm & Conditions I can not get it to work.

Can someone shine a light on this (probably coding error at my side)??

Code part that registers for Server Alerts and 2 parameters to be monitored (i=2258 (Server time) and progStatus, int giving the machine running status (running, stopped etc).

```
public async void ServerSubscription()
            {
                bool subscribed = false;
                while (!subscribed)
                {
                    try
                    {
                        var subscription = new Subscription(session.DefaultSubscription)
                        {
                            PublishingInterval = 3000
                        };
                        Console.WriteLine("subscription created");
                        var selectClause = new[]
                        {
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventId") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventType") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("SourceName") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Time") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Message") }, AttributeId = Attributes.Value },
                            new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Severity") }, AttributeId = Attributes.Value }
                        };
                        //adding items to moniter into the subscription
                        var list = new List<MonitoredItem>
                    {
                        new MonitoredItem(subscription.DefaultItem)
                        {
                            StartNodeId = new NodeId("i=2258"),
                            AttributeId = Attributes.Value,
                            DisplayName = "Server Time",
                            MonitoringMode = MonitoringMode.Reporting,
                            SamplingInterval = 1000,
                            QueueSize = 0,
                            DiscardOldest = true
                        },
                        new MonitoredItem(subscription.DefaultItem)
                        {
                            StartNodeId = new NodeId("ns=2;s=/Channel/State/progStatus"),
                            AttributeId = Attributes.Value,
                            DisplayName = "Program Status",
                            MonitoringMode = MonitoringMode.Reporting,
                            SamplingInterval = 1000,
                            QueueSize = 0,
                            DiscardOldest = true
                        },
                        new MonitoredItem(subscription.DefaultItem)
                        {
                            NodeClass = NodeClass.Object,
                            StartNodeId = ObjectIds.Server,
                            AttributeId = Attributes.EventNotifier,
                            DisplayName = "Server Alerts",
                            MonitoringMode = MonitoringMode.Reporting,
                            SamplingInterval = -1,
                            QueueSize = 100,
                            CacheQueueSize = 100,
                            Filter=new EventFilter() { SelectClauses = selectClause}
                            }
                        };
                        // Console.WriteLine("Moniterd Items are added to the list");
                        list.ForEach(i => i.Notification += OnNotification);
                        session.AddSubscription(subscription);
                        subscription.Create();
                        subscription.AddItems(list);
                        subscription.ApplyChanges();
                        subscription.ConditionRefresh();
                        subscribed = true;
                    }
                    catch (Exception ex)
                    {
                        Console.WriteLine($"{DateTime.Now} [{_machineConfiguration.MachineId}] ! Could not subscribe, retry in 1 min");
                        Console.WriteLine($"{ex.Message}");
                        Console.WriteLine($"{ex.StackTrace}");
                        Thread.Sleep(60000);
                    }
                }
                var timer = new System.Timers.Timer(60 * 1000);
                timer.Elapsed += (sender, e) => PollMachine();
                timer.Start();
            }
```

```
public void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
            {
                foreach (var value in item.DequeueEvents())
                {
                    Console.WriteLine($"{3}: {2}, {4}, {5}", value.EventFields.Select(v => v.Value ?? "<null>").ToArray());
                    Console.WriteLine($"Event Detected {value.EventFields.Select(v => v.Value ?? " < null > ").ToArray()}");
                    EventNotification(item, e);
                }
                foreach (var value in item.DequeueValues())
                {
                    if (item.DisplayName == "Program Status")
                    {
                        Console.WriteLine($"{DateTime.Now} {_machineConfiguration.MachineId} =>Program Status changed event reported by Server: {0}: {1}, {2}, {3}", item.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode);
                        ProgStatusDataChange();
                    }
                }
            }
```

2nd part is the OnNotification part, for the item.DequeueValues(), I receive the values from the 2 parameters. There is nothing being processed at the item.DequeueEvents() part, where I would expect the Alarm & Conditions to be shown....

Thanks in advance!!

Peter

MD-V, 2020-03-27:
@KLMInnovation are you sure that the Alarms&Events are propagated up through the address space to the Server Node ?

KLMInnovation, 2020-03-27:
Hello MD-V,

I used the code from an example that I found on this forum.
I changed the StartNodeId to "ns=2;s=Sinumerik" and now the events are coming.

Thanks, it works now!!

closed 2020-03-27 by KLMInnovation