OPCFoundation/UA-.NETStandard #933: Building UA-.NETStandard with VS2017 or VS2019

mrahman252003, 2020-03-30:
I tried to build the OPC stack by VS2017 or VS2019   in WIN10, in both cases, I get lots of errors.
I had  .net 2.6 installed.  Any idea of how to compile and test the OPC Server?  Thanks in advance.

mregen, 2020-04-01:
Hi @mrahman252003, either one should be fine. Maybe you miss a .Net Core framework or VS C# component. Typically open solution and build. All ci builds pass.

mrahman252003, 2020-04-01:
Hi, mregen, thank you. Is there any list and versions of those needed components anywhere, so I can cross-check with mine one? What do you have in the CI build machine?

AlinMoldovean, 2020-04-02:
Hi @mrahman252003 ,

Make sure you have you have installed .NET Core SDK 2.0 and .NET Framework 4.6.1.
These are the basic prerequisites for VS 2017 and you can get them from here:
https://dotnet.microsoft.com/download/visual-studio-sdks

mrahman252003, 2020-04-04:
Sorry, I tried with .NET Core SDK -2.1.202 (x64) +.NET Framework 4.6.1
I installed VS 2017 community Ver in WIN 10 laptop.  I see lots of missing packages, tried a few to install and it goes messy!

mrahman252003, 2020-04-13:
For me, the solution was clearing all Nuget cache(s) and added Name: nuget.org Source: https://api.nuget.org/v3/index.json
For some reason, when I installed visual studio, it did not add or maybe I need to add (but I didn't know it).

https://stackoverflow.com/questions/52376567/how-to-resolve-unable-to-find-package-nuget-error

closed 2020-04-13 by mrahman252003