OPCFoundation/UA-.NETStandard #936: PLC Sim Advanced

tronghuybku, 2020-04-05:
Hi all. I built OPC UA Server in Tia Portal S7-1500 and PLC SimAdvanced. I want to use OPC.UA.SampleClient to connect My Server but Issue. The client cannot find endpointUrl. I set      <AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates> to connect OPC Server with No security. Prosys OPC UA Server is Ok but OPC UA in S7-1500 cannot connect

fchaxel, 2020-04-21:
Hi,

I cannot test it .. stay at home ! but I have a code working with real S7-1500 where SHA1 is used for the X509 certificate signature. It's rejected by default with this code.

trys 

`config.SecurityConfiguration.RejectSHA1SignedCertificates = false;`

Bye

SebastianBaeumler, 2020-04-21:
Hello, 

do you have some more information (Used PLC, PLC Sim Advenced Version, TIA Portal Version)?

Did you enabled OPC UA Server within the hardware configuration?

[image]

I tried on my Setup with an new project and the connection was successful.

Here you can see some pictures from the connection establishment.
[image]

[image]

closed 2020-05-14 by AlinMoldovean