OPCFoundation/UA-.NETStandard #940: Issue with ComplexTypes with Duration fields.

eoursel, 2020-04-07:
Hi Martin 

I am testing again the ComplexTypesClient on my server (endpoint provided in private email).  

There is an error when trying to get the type Duration (i=290) subtype of Double (i=11). It is not in type dictionary. TryGetValue fails 

 ```
   private Type GetFieldType(StructureField field)
        {
            Type fieldType = null;
            Type collectionType = null;
            if (field.DataType.NamespaceIndex == 0)
            {
                fieldType = Opc.Ua.TypeInfo.GetSystemType(field.DataType, m_session.Factory);
```

       fieldType is Null if field.DataType == "i=290" Duration
Regards

eoursel, 2020-05-03:
hi @mregen 

In order to solve this issue i just added one line in TypeInfo.cs into GetSystemType to add the Duration in the switch. Tell me if this fix is valid. 

Regards

```
public static Type GetSystemType(NodeId datatypeId, EncodeableFactory factory)
        {
            if (datatypeId == null)
            {
                return null;
            }

            if (datatypeId.NamespaceIndex != 0 || datatypeId.IdType != Opc.Ua.IdType.Numeric)
            {
                return factory.GetSystemType(datatypeId);
            }

            switch ((uint)datatypeId.Identifier)
            {
                case DataTypes.Boolean: { return typeof(Boolean); }
                case DataTypes.SByte: { return typeof(SByte); }
                case DataTypes.Byte: { return typeof(Byte); }
                case DataTypes.Int16: { return typeof(Int16); }
                case DataTypes.UInt16: { return typeof(UInt16); }
                case DataTypes.Int32: { return typeof(Int32); }
                case DataTypes.UInt32: { return typeof(UInt32); }
                case DataTypes.Int64: { return typeof(Int64); }
                case DataTypes.UInt64: { return typeof(UInt64); }
                case DataTypes.Float: { return typeof(Single); }
                case DataTypes.Double: { return typeof(Double); }
                case DataTypes.String: { return typeof(String); }
                case DataTypes.DateTime: { return typeof(DateTime); }
                case DataTypes.Guid: { return typeof(Uuid); }
                case DataTypes.ByteString: { return typeof(Byte[]); }
                case DataTypes.XmlElement: { return typeof(XmlElement); }
                case DataTypes.NodeId: { return typeof(NodeId); }
                case DataTypes.ExpandedNodeId: { return typeof(ExpandedNodeId); }
                case DataTypes.StatusCode: { return typeof(StatusCode); }
                case DataTypes.DiagnosticInfo: { return typeof(DiagnosticInfo); }
                case DataTypes.QualifiedName: { return typeof(QualifiedName); }
                case DataTypes.LocalizedText: { return typeof(LocalizedText); }
                case DataTypes.DataValue: { return typeof(DataValue); }
                case DataTypes.BaseDataType: { return typeof(Variant); }
                case DataTypes.Structure: { return typeof(ExtensionObject); }
                case DataTypes.Number: { return typeof(Variant); }
                case DataTypes.Integer: { return typeof(Variant); }
                case DataTypes.UInteger: { return typeof(Variant); }
                case DataTypes.UtcTime: { return typeof(DateTime); }
                case DataTypes.Enumeration: { return typeof(Int32); }
                /* EOL */
                case DataTypes.Duration: { return typeof(Double);}
            }

            return factory.GetSystemType(datatypeId);
        }
```

mregen, 2020-05-04:
Hi @eoursel, in theory the complex type resolver should find the Duration supertype, aka Double. Adding this to the TypeInfo class may have side effects, as Duration is not a BuiltInType, right? 
So I need to double check, but it is definetely a shortcut compared to browsing the supertype. I hopr to get to the issue ver soon, please stay tuned.

Cheers,
Martin

PR #978 (2020-05-15): Complex types fixes
fix #940 and #975 
- add namespace qualifier to complex type name
- improve handling of known subtypes
- fix Pss detection on .Net 462
- stronger check of valid DataTypeDefinition for servers which expose invalid structure (nodejs)
commit f53f71f8 (2020-05-15): Complex types fixes (#978)

fix #940 and #975 
- add namespace qualifier to complex type name
- improve handling of known subtypes
- fix Pss detection on .Net 462
- stronger check of valid DataTypeDefinition for servers which expose invalid structure (nodejs)
closed 2020-05-15 by mregen