OPCFoundation/UA-.NETStandard #946: Creating Subscriptions/MonitoredItems while server state is NoConfiguration

romanzotter, 2020-04-15:
I have following problem/question:
I use the OPC UA client from version 1.4.359.31.

I have an established connection to an OPC UA Server with one Subscription with some MonitoredItems.
When the OPC UA server gets restarted, it allows connections from OPC UA clients, although it has the ServerState “NoConfiguration”.
When the client reestablishes its connection, the Subscription with its MonitoredItem values also gets reconnected.
In this case, the MonitoredItems have the state BadNodeIdUnknown, because the ServerState is “NoConfiguration”.
A few seconds later, the ServerState changes to “Running” and the Nodes would be available (can be seen over an OPC UA-Browser, or when I restart my OPC UA client).
Unfortunately the states of the MonitoredItems stay at BadNodeIdUnknown and don’t get automatically reconnected/updated.

My question here is: 
What can I do in that case? How should the OPC UA client behave when this happens?
Is this or can this be handled by the framework itself?

AlinMoldovean, 2020-04-15:
Hi @romanzotter ,

This seems to be an issue on the server side. It should not process service calls if it is not in the running state to avoid situations like you described. 
The server library of this SDK returns _BadServerHalted_ in case it is not in the complete running state.

On the client side you can implement a custom reconnect procedure to check ServerState or some nodes before recreating the subscriptions and monitored items but this should normally be solved on the server side.

romanzotter, 2020-04-15:
Hi @AlinMoldovean 

So, that I understand this correctly:
If the server is in the State "NoConfiguration", normally it should allow connections from clients, but not allow new subscriptions?

If the server does process the subscription requests while it is not in the state "Running", is it better to disconnect/reconnect the subscription, when it switches to the state "Running"?
Or shall I fully delete the subscription on the client then?

closed 2020-04-22 by romanzotter