OPCFoundation/UA-.NETStandard #953: Events not notified when using StartNodeId != ObjectIds.Server

Tiraflo, 2020-04-23:
Dear All,

I would like to receive events from a particular subset of the address space. So I started by modifying the boiler sample (BoilerNodeManager.cs) in order to send events periodically:

        private void DoSimulation(object state)
        {
            try
            {
                double value1 = m_boiler1.Drum.LevelIndicator.Output.Value;
                value1 = ((int)(++value1))%100;
                m_boiler1.Drum.LevelIndicator.Output.Value = value1;
                m_boiler1.ClearChangeMasks(SystemContext, true);

                double value2 = m_boiler2.Drum.LevelIndicator.Output.Value;
                value2 = ((int)(++value2))%20;
                m_boiler2.Drum.LevelIndicator.Output.Value = value2;
                m_boiler2.ClearChangeMasks(SystemContext, true);

                /* --- Added sources based on AlarmConditionNodeManager.cs --- */
                SystemEventState e = new SystemEventState(m_boiler1.Drum.LevelIndicator);

                e.Initialize(
                    SystemContext,
                    m_boiler1.Drum.LevelIndicator,
                    EventSeverity.Medium,
                    new LocalizedText("Raising Events Boiler 1"));

                e.SetChildValue(SystemContext, Opc.Ua.BrowseNames.SourceNode, m_boiler1.Drum.LevelIndicator.NodeId, false);
                e.SetChildValue(SystemContext, Opc.Ua.BrowseNames.SourceName, m_boiler1.Drum.LevelIndicator.DisplayName, false);

                Server.ReportEvent(e);
                /* --- */
            }
            catch (Exception e)
            {
                Utils.Trace(e, "Unexpected error during simulation.");
            }
        }

Then, on client's side, I create a monitored item according to the Boiler1/Drum/LevelIndicator object (after having browsed through nodes and found this particular one):

	//Console.WriteLine("Subscribe to event for node {0}", ExpandedNodeId.ToNodeId(nextRd.NodeId, session.NamespaceUris));
	Console.WriteLine("Subscribe to event for node {0}", nextRd.NodeId.ToString());
	MonitoredItem monitoredItem = new MonitoredItem(subscription.DefaultItem) {
		DisplayName = nextRd.DisplayName.ToString(),
		//StartNodeId = ObjectIds.Server,
		StartNodeId = nextRd.NodeId.ToString(), // ExpandedNodeId.ToNodeId(nextRd.NodeId, session.NamespaceUris), 
		AttributeId = Attributes.EventNotifier
	};
	monitoredItems.Add(monitoredItem);
	monitoredItem.NodeClass = NodeClass.Variable | NodeClass.Object;
	monitoredItem.SamplingInterval = -1; // 0 ?
	monitoredItem.QueueSize = 100;
	monitoredItem.CacheQueueSize = 100;

The WriteLine call gives the **following** output:

 _Boiler #1, 2:Boiler #1, Object (ns=2;i=138)
'   + Pipe1001, 2:PipeX001, Object (ns=2;i=139)
'   +   +   + FTX001, 2:FTX001, Object (ns=2;i=140)
'   +   +   +   + Output, 2:Output, Variable (ns=2;i=141)
'   +   +   +   +   + EURange, EURange, Variable (ns=2;i=144)
'   +   +   + ValveX001, 2:ValveX001, Object (ns=2;i=147)
'   +   +   +   + Input, 2:Input, Variable (ns=2;i=148)
'   +   +   +   +   + EURange, EURange, Variable (ns=2;i=151)
'   +   +   + FTX001, 2:FTX001, Object (ns=2;i=140)
'   +   +   +   + Output, 2:Output, Variable (ns=2;i=141)
'   +   +   +   +   + EURange, EURange, Variable (ns=2;i=144)
'   + Drum1001, 2:DrumX001, Object (ns=2;i=154)
'   +   +   + LIX001, 2:LIX001, Object (ns=2;i=155)
**Subscribe to event for node ns=2;i=155**
'   +   +   +   + Output, 2:Output, Variable (ns=2;i=156)
'   +   +   +   +   + EURange, EURange, Variable (ns=2;i=159)
'   +   +   + LIX001, 2:LIX001, Object (ns=2;i=155)_

When setting StartNodeId to ObjectIds.Server, then I obtain this kind of notification:

_LIX001: ByteString, System.Byte[]
LIX001: NodeId, i=2130
LIX001: NodeId, ns=2;i=155
LIX001: LocalizedText, LIX001
LIX001: DateTime, 23/04/2020 14:25:17
LIX001: DateTime, 23/04/2020 14:25:17
LIX001: Null,
LIX001: LocalizedText, Raising Events Boiler 1
LIX001: UInt16, 500_

However, when setting StartNodeId to the desired node (ExpandedNodeId.ToNodeId(nextRd.NodeId, session.NamespaceUris or just nextRd.NodeId.ToString()) then I don't receive any notication anymore.

For info:

[image]

What did I misunderstand? What could be missing when Boiler 1/Drum/LevelIndicator is selected as the starting node?

Regards.

11v1, 2020-04-26:
You calls ReportEvent on Server:
`Server.ReportEvent(e);`
From your server code it is not seen what variable "Server" is, but I suppose it is Server Node. If it so then that is the answer: you will receive events from the node that reports it or a "parent" node.

Tiraflo, 2020-04-27:
Indeed, that's the server node (which I didn't noticed when "copying/pasting" from different samples...) and by replacing with the appropriate node (Boiler/Drum/LevelIndicator), I can now receive events when subscribing to this particular node.

In addition, if I properly understand, I can effectively receive events by subscribing to node's parent because the EventNotifier attribute and the HasNotifier reference are defined to allow it. Also, I can imagine that an event could have a SourceNode attribute which does not necessary match the node which report this event.

Thank you for your help.

closed 2020-05-14 by AlinMoldovean