OPCFoundation/UA-.NETStandard #958: Rejected Certificate "NetCoreConsoleClient"

torsilver, 2020-04-28:
.Net Core OPC UA Console Client sample
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://win-pmrlpa42nhm:4845/.
    Selected endpoint uses: Basic128Rsa15
3 - Create a session with OPC UA server.
Rejected Certificate: DC=WIN-PMRLPA42NHM, CN=OPC.SimaticNET.S7
Exception: One or more errors occurred. (Certificate is not trusted.
SubjectName: DC=WIN-PMRLPA42NHM, CN=OPC.SimaticNET.S7
IssuerName: DC=WIN-PMRLPA42NHM, CN=OPC.SimaticNET.S7)

<AutoAcceptUntrustedCertificates>autoAccept</AutoAcceptUntrustedCertificates> is like this.
But "if (config.SecurityConfiguration.AutoAcceptUntrustedCertificates)"is still flase.
so autoAccept is still flase.
if I set autoAccept = true.

3 - Create a session with OPC UA server.
Accepted Certificate: DC=WIN-PMRLPA42NHM, CN=OPC.SimaticNET.S7
Exception: One or more errors occurred. (BadSecureChannelClosed)

Is there any settings I should change?

I am trying to connect with another compter with "Siemens Automation" V8.1

AlinMoldovean, 2020-04-28:
Hi @torsilver ,

It seems the UA Client certificate is rejected on server side.
You need to check on server side how to accept the certificate.

torsilver, 2020-04-28:
Ok thanks. I will do that.

On 04/28/2020 16:44, Alin Moldovean wrote:

Hi @torsilver ,

It seems the UA Client certificate is rejected on server side.
You need to check on server side how to accept the certificate.

torsilver, 2020-04-29:
> Hi @torsilver ,
> 
> It seems the UA Client certificate is rejected on server side.
> You need to check there how to accept the certificate.

Thanks a lot. It works now.
In the "Siemens" software. I just accept my new client. And it works now.
Thanks a lot.

closed 2020-04-29 by torsilver