OPCFoundation/UA-.NETStandard #962: ReflectionTypeLoadException: Exception of type "System.Reflection.RefectionTypeLoadException" was thrown

letruongthanh24103698, 2020-04-30:
Hi everyone. 
I'm developing an OPC UA CLIENT for Android in Unity3d.
here's my steps:
1. In vs2017 create **empty console app** set .NET Framwork to 4.6.
2. Download OPCFoundation .NETSTANDARD in Nuget manager.
3. Make release without any lines of code.
4. go to folder release, copy all dll file and put it in Asset/Plugins/OPC>NET of my Android Unity3d project.
5. create a gameobject in unity with this code (rewrite from NetConsoleSampleClient):
`using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Opc.Ua;
using Opc.Ua.Client;
using Opc.Ua.Configuration;
using System;

public class OPCUA : MonoBehaviour
{
    // Start is called before the first frame update
    static bool autoAccept = true;
    //static Session session;
    static string endpointURL = "opc.tcp://localhost:49320/";

    void Start()
    {
        Connect();
    }

    static async void Connect()
    {
        Debug.Log("Step 1 - Create a config.");
        //var config = new ApplicationConfiguration()
        ApplicationInstance application = new ApplicationInstance
        {
            ApplicationName = "test-opc",
            ApplicationType = ApplicationType.Client,
            //SecurityConfiguration = new SecurityConfiguration { ApplicationCertificate = new CertificateIdentifier() },
            //TransportConfigurations = new TransportConfigurationCollection(),
            //TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
            //ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000 }
            ConfigSectionName = "Opc.Ua.MonoSampleClient"
        };
        //config.Validate(ApplicationType.Client);
        // load the application configuration.
        ApplicationConfiguration config = await application.LoadApplicationConfiguration("Assets/Plugins/Opc.Ua.MonoSampleClient.Config.xml",false);

        // check the application certificate.
        bool haveAppCertificate = await application.CheckApplicationInstanceCertificate(false, 0);

        if (haveAppCertificate)
        {
            config.ApplicationUri = Utils.GetApplicationUriFromCertificate(config.SecurityConfiguration.ApplicationCertificate.Certificate);
            if (config.SecurityConfiguration.AutoAcceptUntrustedCertificates)
            {
                autoAccept = true;
            }
            config.CertificateValidator.CertificateValidation += new CertificateValidationEventHandler(CertificateValidator_CertificateValidation);
        }
        else
        {
            Debug.Log("    WARN: missing application certificate, using unsecure connection.");
        }

        Debug.Log(config.ApplicationUri);

        /*if (config.SecurityConfiguration.AutoAcceptUntrustedCertificates)
        {
            config.CertificateValidator.CertificateValidation += (s, e) => { e.Accept = (e.Error.StatusCode == StatusCodes.BadCertificateUntrusted); };
            //config.CertificateValidator.CertificateValidation += (s, e) => { e.Accept = true; };
        }  */

        Debug.Log("Step 2 - Create a session with your server.");
        //var selectedEndpoint = CoreClientUtils.SelectEndpoint(endpointURL, haveAppCertificate, 15000);
        /*var selectedEndpoint = CoreClientUtils.SelectEndpoint(endpointURL, haveAppCertificate, 15000);

        Debug.Log("    Selected endpoint uses: "+
           selectedEndpoint.SecurityPolicyUri.Substring(selectedEndpoint.SecurityPolicyUri.LastIndexOf('#') + 1));

        Debug.Log("3 - Create a session with OPC UA server.");
        var endpointConfiguration = EndpointConfiguration.Create(config);
        var endpoint = new ConfiguredEndpoint(null, selectedEndpoint, endpointConfiguration);*/

        var session = await Session.Create(config, new ConfiguredEndpoint(null, new EndpointDescription("opc.tcp://localhost:49320/")), false, "OPC UA Debug Client", 60000, new UserIdentity(new AnonymousIdentityToken()), null);
        Debug.Log("Step 3 - Read Node");
        while (true)
        {
            DataValue tmp = session.ReadValue("ns=2;s=Channel1.Device1.Tag2");
            Debug.Log(tmp);

        }
    }

    // Update is called once per frame
    void Update()
    {
        
    }

    private static void CertificateValidator_CertificateValidation(CertificateValidator validator, CertificateValidationEventArgs e)
    {
        if (e.Error.StatusCode == StatusCodes.BadCertificateUntrusted)
        {
            e.Accept = autoAccept;
            if (autoAccept)
            {
                Debug.Log("Accepted Certificate: "+ e.Certificate.Subject);
            }
            else
            {
                Debug.Log("Rejected Certificate: " +e.Certificate.Subject);
            }
        }
    }
   
}
`
Then I run, It cannot run to "Step 3 - Read Node" and get 3 error:
ERR1
ReflectionTypeLoadException: Exception of type 'System.Reflection.ReflectionTypeLoadException' was thrown.
System.Reflection.Assembly.GetExportedTypes () (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
Opc.Ua.EncodeableFactory.AddEncodeableTypes (System.Reflection.Assembly assembly) (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Opc.Ua.EncodeableFactory..ctor () (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Opc.Ua.EncodeableFactory..cctor () (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Rethrow as TypeInitializationException: The type initializer for 'Opc.Ua.EncodeableFactory' threw an exception.
Opc.Ua.ServiceMessageContext..ctor () (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Opc.Ua.ServiceMessageContext..ctor (System.Boolean shared) (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Opc.Ua.ServiceMessageContext..cctor () (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Rethrow as TypeInitializationException: The type initializer for 'Opc.Ua.ServiceMessageContext' threw an exception.
Opc.Ua.ApplicationConfiguration.CreateMessageContext (System.Boolean clonedFactory) (at <7eb6d5e10b2f4a2f82a6a88d1cb5c1ff>:0)
Opc.Ua.Client.Session+<Create>d__77.MoveNext () (at <251a9d3b07734b08819f0b476aeea2dd>:0)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
Opc.Ua.Client.Session+<Create>d__76.MoveNext () (at <251a9d3b07734b08819f0b476aeea2dd>:0)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
OPCUA+<Connect>d__3.MoveNext () (at Assets/OPCUA.cs:77)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <e1319b7195c343e79b385cd3aa43f5dc>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at C:/buildslave/unity/build/Runtime/Export/UnitySynchronizationContext.cs:155)
UnityEngine.UnitySynchronizationContext:ExecuteTasks() (at C:/buildslave/unity/build/Runtime/Export/UnitySynchronizationContext.cs:105)
ERR2
Unloading broken assembly Assets/Plugins/OPC.NET/Microsoft.AspNetCore.WebUtilities.dll, this assembly can cause crashes in the runtime
ERR3
Unloading broken assembly Assets/Plugins/OPC.NET/Opc.Ua.Core.dll, this assembly can cause crashes in the runtime

I have try this code in Console APP (vs2017) before, and it run Ok.
Any idea to fix this. Or please show me how to build an Android Unity3d project with OPC UA.

Thanks for reading. (This's too long. Sorry)

MD-V, 2020-04-30:
@letruongthanh24103698 See this [Discussion](https://github.com/OPCFoundation/UA-.NETStandard/issues/760#issuecomment-526571741)

for help with unity.

letruongthanh24103698, 2020-04-30:
> @letruongthanh24103698
> 
> You will have to create a link.xml to prevent Unity from stripping the ByteCode.
> See this [Discussion](https://github.com/OPCFoundation/UA-.NETStandard/issues/760#issuecomment-526571741) for more information.
a link.xml file has been created and put in Assets folder. but does not work

letruongthanh24103698, 2020-04-30:
> @letruongthanh24103698
> 
> You will have to create a link.xml to prevent Unity from stripping the ByteCode.
> See this [Discussion](https://github.com/OPCFoundation/UA-.NETStandard/issues/760#issuecomment-526571741) for more information.

letruongthanh24103698, 2020-04-30:
i found solution. when copy dll file, i must copy their xml file.

closed 2020-04-30 by letruongthanh24103698