OPCFoundation/UA-.NETStandard #967: Vairable Data Node with Historical Configauration

cannonkalra, 2020-05-06:
I've been trying to make a node that has historical data with the following code

            BaseDataVariableState node = new BaseDataVariableState(null);
            node.TypeDefinitionId = VariableTypeIds.DataItemType;
            node.SymbolicName = name;
            node.NodeId = new NodeId(path, NamespaceIndex);
            node.BrowseName = new QualifiedName(name, NamespaceIndex);
            node.DisplayName = new LocalizedText("en", name);
            node.Description = null;
            node.WriteMask = 0;
            node.UserWriteMask = 0;
            node.DataType = DataTypeIds.Double;
            node.ValueRank = ValueRanks.Scalar;
            node.AccessLevel = AccessLevels.HistoryReadOrWrite | AccessLevels.CurrentRead;
            node.UserAccessLevel = AccessLevels.HistoryReadOrWrite | AccessLevels.CurrentRead;
            node.MinimumSamplingInterval = MinimumSamplingIntervals.Indeterminate;
            node.Historizing = true;

            PropertyState<Annotation> annotation = new PropertyState<Annotation>(node);
            annotation.ReferenceTypeId = ReferenceTypeIds.HasProperty;
            annotation.TypeDefinitionId = VariableTypeIds.PropertyType;
            annotation.SymbolicName = Opc.Ua.BrowseNames.Annotations;
            annotation.BrowseName = Opc.Ua.BrowseNames.Annotations;
            annotation.DisplayName = new LocalizedText("en", name);
            annotation.Description = null;
            annotation.WriteMask = 0;
            annotation.UserWriteMask = 0;
            annotation.DataType = DataTypeIds.Annotation;
            annotation.ValueRank = ValueRanks.Scalar;
            annotation.AccessLevel = AccessLevels.HistoryReadOrWrite;
            annotation.UserAccessLevel = AccessLevels.HistoryReadOrWrite;
            annotation.MinimumSamplingInterval = MinimumSamplingIntervals.Indeterminate;
            annotation.Historizing = false;
            node.AddChild(annotation);

            annotation.NodeId = new NodeId(name + '/' + Opc.Ua.BrowseNames.Annotation, NamespaceIndex);// NodeTypes.ConstructIdForComponent(m_annotations, namespaceIndex);

            HistoricalDataConfigurationState configuration = new HistoricalDataConfigurationState(node);
            configuration.MaxTimeInterval = new PropertyState<double>(configuration);
            configuration.MinTimeInterval = new PropertyState<double>(configuration);
            configuration.NodeId = new NodeId(name + '/' + Opc.Ua.BrowseNames.HAConfiguration, NamespaceIndex);
            configuration.BrowseName = Opc.Ua.BrowseNames.HAConfiguration;
            configuration.DisplayName = Opc.Ua.BrowseNames.HAConfiguration;
            configuration.ReferenceTypeId = ReferenceTypeIds.HasHistoricalConfiguration;
            configuration.TypeDefinitionId = ObjectTypeIds.HistoricalDataConfigurationType;

            configuration.Create(
               SystemContext,
               configuration.NodeId,
               configuration.BrowseName,
               configuration.DisplayName,
               true);

            configuration.StartOfArchive.Value = DateTime.Now;
            configuration.StartOfOnlineArchive.Value = DateTime.UtcNow;

            if (parent != null)
            {
                parent.AddChild(node);
            }

            return node;
}

When try to find this variable on client, and check it's HA Configuration for StartOfArchive, I find Timestamps have updated but the value has not, for which I receive an Exception from the HIstorical Access Client 

[image]

On checking with Prosys UA Browser, I found the Value inside Value of PropertyState StartOfArchive was not changing in my case, but was fine in bottom screen shot (from HistoricalAccessServer Example)

[image]

This is from the Reference Server from the Workshop
[image]

cannonkalra, 2020-05-07:
On inspecting, the issue is with writing PropertyState.. I'm not able to write change values under HA Configuration.. after calling Create function..

mrsuciu, 2023-09-27:
Please reopen if still needed

closed 2023-09-27 by mrsuciu